sinara-hw/Urukul #1: SYSU error with internal clock

jbqubit, 2018-04-19:
I'm running the SYSU Kasli variant with EEMs TTL and Urukul. Running gateware and firmware version d0f6123f. Relying on internal Kasli internal clock I get an error. 

```
device_db = {
    "core": {
        "type": "local",
        "module": "artiq.coredevice.core",
        "class": "Core",
        "arguments": {
            "host": core_addr, 
            "ref_period": 1e-9
            #,"external_clock": True
            # ,"ref_multiplier": 4
        }
    },
...
    urukul0_cpld={
        "type": "local",
        "module": "artiq.coredevice.urukul",
        "class": "CPLD",
        "arguments": {
            "spi_device": "spi_urukul0",
            "io_update_device": "ttl_urukul0_io_update"
            ,"refclk": 125e6
            ,"clk_sel": 0
        }
    }
```

```
$ artiq_run repository/demo.py 
/home/britton/miniconda3/envs/artiq-dev/lib/python3.5/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
Traceback (most recent call last):
  File "/home/britton/artiq-dev/artiq/artiq/master/worker_db.py", line 159, in get
    dev = _create_device(desc, self)
  File "/home/britton/artiq-dev/artiq/artiq/master/worker_db.py", line 97, in _create_device
    return device_class(device_mgr, **desc.get("arguments", {}))
  File "/home/britton/artiq-dev/artiq/artiq/coredevice/ad9910.py", line 69, in __init__
    assert sysclk <= 1e9
AssertionError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/britton/miniconda3/envs/artiq-dev/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/britton/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/britton/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/britton/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 196, in run
    exp_inst = _build_experiment(device_mgr, dataset_mgr, args)
  File "/home/britton/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 182, in _build_experiment
    return get_experiment(module, args.experiment)(managers)
  File "/home/britton/artiq-dev/artiq/artiq/language/environment.py", line 216, in __init__
    self.build(*args, **kwargs)
  File "repository/demo.py", line 8, in build
    self.urukul0s = [self.get_device("urukul0_ch"+str(i)) for i in range(4)]
  File "repository/demo.py", line 8, in <listcomp>
    self.urukul0s = [self.get_device("urukul0_ch"+str(i)) for i in range(4)]
  File "/home/britton/artiq-dev/artiq/artiq/language/environment.py", line 276, in get_device
    return self.__device_mgr.get(key)
  File "/home/britton/artiq-dev/artiq/artiq/master/worker_db.py", line 162, in get
    .format(name)) from e
artiq.master.worker_db.DeviceError: Failed to create device 'urukul0_ch0'
```

jbqubit, 2018-04-19:
```
from artiq.experiment import *

class UrukulTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul0_cpld")
        self.urukul0s = [self.get_device("urukul0_ch"+str(i)) for i in range(4)]
        self.ttl = self.get_device("ttl0")

    @kernel
    def run(self):
        f = 3.0*MHz
        dt = 2*us

        self.core.reset()
        self.ttl.output()
        delay(1*ms)

        for j in range(4):
            self.urukul0s[j].init()

        while True:
            with parallel:
                self.urukul0s[0].set(f, 0., 1.)
                self.urukul0s[1].set(f, 0., 1.)
                self.ttl.pulse(dt)
                self.urukul0s[0].sw.pulse(dt)
                self.urukul0s[1].sw.pulse(dt)

                # for j in range(4):
                #     self.urukul0s[j].sw.pulse(2*us)
            delay(1*ms)
```

hartytp, 2018-07-30:
@jbqubit AFAICT, this is an ARTIQ issue, not a HW issue.

closed 2018-07-30 by hartytp