sinara-hw/Urukul #3: Urukul v1.0 detailed testing

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-07 20:29](https://github.com/sinara-hw/sinara/issues/354)_

- [x] Channel-to-channel cross-talk: adjacent channels, with agressor, victim switches opened and switches closed and with attenuators at max/min
- [x] Output phase noise/amplitude noise (0.1 Hz - 10 MHz)
- [x] OIP3, OIP2 vs frequency for min attenuator setting and medium/large amplitude (ASF)
- [x] RF switch settling behavior for on edge at max power (worst case)
- [x] Attenuator glitches
- [x] Nyquist attenuation, clock leakage
- [ ] RF switch settling behavior for off edge at max power (worst case)
- [ ] RF switch phase chirp
- [ ] Temperature coefficient of power and phase
- [ ] Are we happy running the IC11 and running the LVDS interface directly from the 3V6 rail? (Try this and check performance)
- [ ] Check performance of clock buffers when running in LVDS mode from the 1V8A rail
- [x] Optimize PLL loop filters (@weidazhang can you do this for the AD9910 boards, please?)
- [ ] Board-to-board crosstalk (in proximity to Kasli, DIO pulsing 5 V into 50 R, Urukul)

Use some fixed reference frequency for all tests, such as 80MHz.

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-07 20:30](https://github.com/sinara-hw/sinara/issues/354#issuecomment-342612378)_

@gkasprow Can you ship us all but 1 of the AD9910 Urukul prototypes you've ordered as soon as basic functionality/"no smoke" testing is complete (i.e. don't complete the above detailed testing first!). 

Thanks!

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-07 20:30](https://github.com/sinara-hw/sinara/issues/354#issuecomment-342612519)_

@gkasprow @jordens Feel free to add more tests to the above list.

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-08 03:32](https://github.com/sinara-hw/sinara/issues/354#issuecomment-342700395)_

@gkasprow When you ship the AD9910 DDSs that we paid for, please can you ship one to @jordens?

marmeladapk, 2018-06-13:
_From @jordens on [2017-11-08 08:47](https://github.com/sinara-hw/sinara/issues/354#issuecomment-342750073)_

Updated it a bit. Please ship right after smoke test:

- [ ] power supplies good
- [ ] clock fanout fans out
- [ ] dds pll locks
- [ ] dds outputs frequency
- [ ] attenuators can be set
- [ ] rf switches work

We'll be happy to help with characterization afterwards.

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-12 18:22](https://github.com/sinara-hw/sinara/issues/354#issuecomment-343756425)_

@gkasprow Is Urukul due to arrive this week? When it arrives, please can we prioritise no smoke/basic functionality testing (see reobert's list) above finishing Zotino -- I'd like to get Urukul prototypes shipped asap. No need to do the detailed testing for now.

Thanks!

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-11-12 18:26](https://github.com/sinara-hw/sinara/issues/354#issuecomment-343756709)_

@hartytp OK, 2 Zotinos + BNC adapters were shipped on Friday. Once I receive DDS, I will test them ASAP.

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-12 18:27](https://github.com/sinara-hw/sinara/issues/354#issuecomment-343756742)_

Thanks!

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-11-12 18:32](https://github.com/sinara-hw/sinara/issues/354#issuecomment-343757125)_

@hartytp There are no panels for DAC BNC adapters attached - I forgotten about them. Will ship them next time with some other stuff.

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-12 18:43](https://github.com/sinara-hw/sinara/issues/354#issuecomment-343757823)_

@gkasprow No problem. That can wait.

If there is any time before Urukul arrives, it'd be great to get a few of the analog test for Zotino done.

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-11-12 19:04](https://github.com/sinara-hw/sinara/issues/354#issuecomment-343759155)_

@hartytp Sure, once I repair my bard, will continue with measurements.

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-12 19:23](https://github.com/sinara-hw/sinara/issues/354#issuecomment-343760366)_

Great, thanks @gkasprow!

marmeladapk, 2018-06-13:
_From @jordens on [2017-11-19 16:38](https://github.com/sinara-hw/sinara/issues/354#issuecomment-345530298)_

@hartytp could you remember to do a bit of triage on the issues you file (i.e. assignee, labels, milestone)?

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-19 18:48](https://github.com/sinara-hw/sinara/issues/354#issuecomment-345539884)_

sure

marmeladapk, 2018-06-13:
_From @hartytp on [2017-11-20 18:55](https://github.com/sinara-hw/sinara/issues/354#issuecomment-345792391)_

@gkasprow For the current source biasing, my inclination is to go for Daniel's suggestion of the LT3092. They're pretty cheap ($2 in quantity) and seem like a better/easier solution than building our own current source.

@jordens @gkasprow if you're oaky with this approach, then we should buy an eval board at some point and test this out with the v1.0 boards we have. I'm happy to do this testing if that helps -- I probably won't do this until after Christmas, but I don't think we're in a rush for it.

marmeladapk, 2018-06-13:
_From @hartytp on [2017-12-11 18:26](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350813087)_

Cool.

Out of curiosity, do you have a proper phase noise/amplitude noise meter? Or, will you just stick this on a spectrum analyser?

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-11 19:16](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350827749)_

We have proper tools.

@gkasprow how are the AD9912 variant boards?

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 21:11](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350859436)_

@jordens I have them all in my lab but was busy with Sayma Ethernet debugging. You can follow my fight [here](https://github.com/m-labs/artiq/issues/854)

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 21:12](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350859672)_

So the next thing is 9912 testing.

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-11 22:32](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350880532)_

I am following that and involved.
Thanks for the AD9912 update. Once you have smoke-tested them like the AD9910 stuff, you can send two of them to me.

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-11 22:45](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350883486)_

@gkasprow what was actually flashed into the CPLD on the 9910 board we have in Berlin?

I see 2 options
https://github.com/m-labs/sinara/tree/master/ARTIQ_ALTIUM/EEMs/Urukul/CPLD_code
or
https://github.com/quartiq/urukul

I am testing h/w but can't generate any output on the IO_SESET (P129) and some pins around.

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-11 23:02](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350887345)_

@gkasprow how you tested 3U DDS 9910?
Have you got DDS ic working?
Do you have the code that sent the spi commands, what actually sent to DDS/atten etc?

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 23:02](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350887492)_

@a-shafir I burned Roberts Verilog code. The only change I did was assignment of test points. Now we have:
```
assign tp = (cfg_data_rst | ((~en_9910) & tstriple7_i));
assign tp_1 = dds_cs_n;
assign tp_2 = dds_sck;
assign tp_3 = dds_sdi;
assign tp_4 = dds_io_update;

```

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 23:05](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350888034)_

I connected 9910 devkit via DIO-Tester module to make sure that Robert's code works. In this way I could communicate with it using LVTTL signals

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 23:06](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350888356)_

I also modified the verilog code so all 4 channels were active simultaneously. In this way I tested if PLLs are locked. Here is the code:
[Urukul_CPLD_all_ch.zip](https://github.com/m-labs/sinara/files/1549674/Urukul_CPLD_all_ch.zip)

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 23:08](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350888731)_

To do general purpose communication over SPI I use little Artix 7 devkit with USB and FORTH processor that translates UART commands to bit-bang protocol, i.e. SPI. It works really fast but it's written in FORTH, so probably not really useful for you :)

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-11 23:13](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350889724)_

@gkasprow i see io_reset commented in the code. probably some other changes.
Please also attach the pin assignment for the code.
So what was generating the DDS SPI command? 9910 devkit? I have no devkit.
Can you generate a dump (sequence) of the commands? 
Also what about the reading from DDS when all in parallel?

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-11 23:15](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350890259)_

@gkasprow i see why in_sel not working to - it commented out
Good that i got this data from you )

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 23:19](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350890994)_

The original DDS was working in parallel. So I didn't read the status from externally connected ones.
You can compare this code with one @jordens posted. I did not touch the pins assignment.
My CPLD project is [here](https://www.dropbox.com/sh/qj16dhavo4qxsw9/AAB5paEl0olZpSaRMjhjxSyOa?dl=0)

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 23:21](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350891401)_

This is how typical setup with devkit looks like. I this case I connected Sampler module, but the idea is same.
[image]

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 23:24](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350892153)_

this is the FORTH code I use to talk to LV595 registers.
The FORTH ascii is executed in FPGA and is toggling IOs routed to an IO register with a few MHz rate. CPU runs with 50MHz clock.
The function below writes 16 bits via SPI to the register and reads them back.

````
: spi_LV595 ( a ) 
	
	1 OUT0_SET_REG io! \ SCK hi
	1 16 lshift OUT0_SET_REG io! \ OE buff active
	1 2 lshift OUT0_CLR_REG io! \ RCLK low
	
    16 0 do
        dup
        1 15 lshift and
        if
           1 1 lshift OUT0_SET_REG io!
        else
           1 1 lshift OUT0_CLR_REG io!
        then
		1 OUT0_CLR_REG io! \ SCK low
		2*
		INP0_REG io@ 4 rshift 1 and +
        1 OUT0_SET_REG io! \ SCK hi	
    loop
	
	 1 2 lshift OUT0_SET_REG io! \ RCLK high
	 1 2 lshift OUT0_CLR_REG io! \ RCLK low
	 1 16 lshift OUT0_CLR_REG io! \ OE buff inactive
	 $ffffff and  \ return only valid bits
;

````

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 23:32](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350893612)_

@a-shafir you are right - I had to modify the reset pin state to make it working.

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-11 23:34](https://github.com/sinara-hw/sinara/issues/354#issuecomment-350894149)_

And the code you have burned in your CPLD is in Urukul_CPLD_1ch.v file

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-12 16:51](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351112532)_

@jordens here is your 9912 :)
[image]

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-12 16:56](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351114267)_

@gkasprow Nice! Other than the missing amplifier, are there any other issues? Is the solder on the underside bad as well?

marmeladapk, 2018-06-13:
_From @hartytp on [2017-12-12 16:59](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351115163)_

Nice!

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-12 17:03](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351116459)_

I got this amp separately. The PN is wrong so they got ones with straight leads and had to replace them.
It seems that the ICs they got had problems with wetting because other chips are fine. Anyway, it's their responsibility to solder it right

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-12 17:15](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351120206)_

Are you returning the AD9912 boards to Technosystem to be fixed as well?

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-12 18:06](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351135089)_

I will fix one piece myself to test it. Rest of the boards will be returned. I talked with the company and they say that it is quite standard thing that some chips have poor wetting and side contacts are not always covered by the tin. The flux that is part of the soldering paste is only roughly 10..15% of it and is not always able to clean the contact side. The joint still maintains the contact because it is low side of it which does the job. Side area does not need to be covered, but if we want, they can fix it.

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-13 08:47](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351323416)_

IMHO it is likely that the joints are bad on the underside as well if they fail to wet that dramatically on the edge. And since they probably didn't inspect those joints there is no telling. I'd like to see them fix it.

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-13 09:09](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351328847)_

I just desoldered the chip and the bottom pads are correctly soldered. Anyway it just looks non professional and should be fixed.

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-13 10:42](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351353248)_

This is a long known for the technologists that too much solder is not helping and even can cover some problem. But in this case it is not about the professionalism but about Optical inspection standards. All the packages designed that way that there is possibility to ensure by inspection the proper electrically and mechanically soldering. For this package there must be wetting on the side edges of the pads. This indicates that there is wetting under the pads and also improves mechanical properties. 

For some of the packages we have (BGA and some of the qfn with signal pads under) the OIS will not enough. I think need to check if the factory can do x-ray inspection, they must have it in or out house at least for BGA but also for some regulator chips etc. Visual inspection of the QFN etc is the must. It shall not increase the cost of the samples production more than few %

Any way investing our time to solving the technological issues is the worst case scenario.

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-13 12:50](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351381968)_

Urukul 9910, 4th channel, 10.000MHz, variable attenuation test, no termination:
[image]

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-13 13:29](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351391245)_

@a-shafir look [here](https://github.com/m-labs/sinara/issues/377#issuecomment-351379129)

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-13 13:40](https://github.com/sinara-hw/sinara/issues/354#issuecomment-351393954)_

@gkasprow ok, this is a news for me too. There are 3 similar looking "QFN-style" packages:
- one have sides plated to be good for wetting (some crystals for instance and old stile QFN)
- some have a etch/step around to have the wetting visible for OIS
- the newest are basicly BGA-like i.e. all wetting is under the package but just no balls.

That minds for us:
- Even we don't have a BGA on a board we need to inspect the packages to ensure they are good for optical inspection (OIS) and if any one of them is needed we must add X-ray inspection into the production list.
- Need to chose the parts for the boards more careful. If the board normally (most of the parts) does not needs X-ray try to avoid any packages that have no OIS compatible pads (like LTM4622 etc) 
- It is good to put this as a part of the h/w and production specs checklist.

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-18 00:41](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352299689)_

Gateware, core drivers and experiment for Urukul AD9910 rev 1.0 board.

The current setup is for KC705, see the details here:
https://github.com/m-labs/artiq/commit/dd3c8a336dc12cda33cadebdf3bafc77c0bc1565

The experiment initializes FMC DIO, the Urukul board and configures the single tone for all 4 channels.
Attenuator and SW are supported.

There are 2 "interference generator loops" (for DDS3 SPI and for ATT SPI). Each can be activated in the experiment. Note: with the current RTIO gateware it is not possible to generate the NU-mode intensive SPI communication so if any interference from the SPI clock will be seen on the output it will be not the worst case scenario.

The 4-channels DDS output:
[image]

The setup:
[image]

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-18 17:40](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352500086)_

* RF Switch rise time is roughly 100ns to 95%
* Spectrum looks ok. A few weird close in Spurs (+- 1.2 kHz and 11kHz at -70dBc) potentially related to grounding to be debugged. They are big when the board is on the antistatic bag on an optical table and otherwise disappear.

More data and screenshots to follow.

marmeladapk, 2018-06-13:
_From @dhslichter on [2017-12-18 18:53](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352523350)_

@a-shafir why are you using scope probes to look at the DDS outputs instead of SMA cables?

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-18 19:01](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352525555)_

@jordens in such setup you easily receive and induce current from every short wave radio station including CB :)

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-18 19:57](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352540600)_

@gkasprow in general yes, but not if you are in a room that is as insanely shielded as the one we are doing the tests in ;)

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-19 11:24](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352720977)_

@dhslichter there was only one more SMA cable in the lab and no 50R terminators at all. Any way the probes are OK to check how the driver controls.

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-19 19:17](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352859041)_

The screenshots are crappy.
tl;dr: All good. The only issue are phase noise spurs at -75dBc at ~1.2 kHz offset with many harmonics, especially x11. We didn't figure out where exactly they came from. The spur offset is surprisingly low. Their offset is temperautre dependent (increases with temperature). And coupling different parts of the PCB (by finger), especially around the underside near the clock fanout, makes them worse. That spur needs to be debugged.

## Setup

* Urukul v1.0 (with BIAS and resistor patches)
* urukul-cpld v1.0rc6
* very clean 100 MHz reference, 10 dBm, un-un, into SMA, selected
* divide-by-4 from clock fanout
* PLL locked, multiplier 40, 387µA, VCO5, 1 GHz VCO
* CFR1=0x00000002, CFR2=0x01400820
* output frequency 80 MHz (0x147ae148) or 81 MHz (0x14bc6a7f) unless otherwise noted
* un-un on output
* using channel 1 unless otherwise noted
* 12 V lab power supply or random 12V 6A brick: no difference

## low frequency limit

* 100 kHz out: -20 dBm max (attenuator at 63 digital, switch open)
* 1 MHz out: 10 dBm

## switch isolation

* 70 dB (attenuator at 63 digital)

## crosstalk

* -84 dB (victim attenuator at 63, switch open, DDS off, aggressor switch open, attenuator at 63)
* -110 dB (same as above but victim switch closed)

## cross-channel intermodulation

* -85 dBc (agggressor 1 MHz, 10 dBm, victim 80.2 MHz, IMD product at 81.2 MHz)
* -90 dBc (agggressor 1.234 MHz, 10 dBm, victim 80.123 MHz, IMD product at 81.357 MHz)
* <-100 dBc (agggressor 80.123 MHz, 10 dBm, victim 81.234 MHz, IMD product at 82.345 MHz)

##  Attenuator step from 20 to 60 digital (16+4dB switch glitch)
[image]
[image]

## Attenuator step from 31 to 32 digital (major carry glitch)
[image]

## Harmonics DC to 1 GHz, clock and ref leakage below RBW
[image]

## Harmonics 2 and 3 vs power

6dBm outputs, 3rd harmonic at -53dB
[image]

10dBm outputs, 3rd harmonic at -25dB
[image]

## Close in spurs (1.2 kHz multiples, especially 11x)

* increase and change the harmonics distribution when touching the underside of the PCB near the clock fanout
* increase when pushing the underside into the antistatic bag (and thereby coupling parts of the underside)
* increase in frequency with increasing temperature
* no power rail or testpoint found where whey originate
* power supply does not make a difference
* shielding by covering with a massive Aluminum case does not change them
* adding more un-uns does not change them
* cable routing does not affect them
* distance to kc705 does not affect them

[image]

## PLL bump and 1 MHz spurs

* spurs at 1 MHz and multiples from the SMPS at -85 dBc on the same level as the specified spurs of the DDS.

[image]

## Phase noise 100 mHz to 10MHz
[image]

## Phase noise 10 Hz to 10 MHz
[image]

## Phase noise 1 kHz to 1MHz
[image]

## RF switch turn on transient
[image]

## Nyquist rejection 400 MHz to 600 MHz
[image]

## Nyquist rejection 450 MHz to 550 MHz
[image]

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-19 19:39](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352864722)_

@jordens such spur may be caused by PSU instability. Then 1.2kHz ripples caused by SMPS or LDO feedback loop may cause such spurs.
So you can have a look with scope on critical power rails.

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-19 19:59](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352869655)_

Ack. We didn't find anything noticable on the rails. We'll do more prodding and poking.

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-19 20:11](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352872838)_

@jordens I can have a look with spectrum analyzer or when necessary with SSA. Is ADI devkit sufficient to recreate this spur?

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-19 21:09](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352886893)_

I don't know. But I'd be surprised if you see them on the devkit.
I'll give you details about the setup later.

marmeladapk, 2018-06-13:
_From @gkasprow on [2017-12-19 21:11](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352887718)_

No, I mean Urukul controlled by ADI devkit instead of the Artiq

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-20 10:06](https://github.com/sinara-hw/sinara/issues/354#issuecomment-353020765)_

I think you should see them on Urukul with your setup. I added the description of the setup to the post above.

marmeladapk, 2018-06-13:
_From @a-shafir on [2017-12-20 13:06](https://github.com/sinara-hw/sinara/issues/354#issuecomment-353058269)_

@jordens what reference clock used for the test? KC705 output from the FPGA pll possible has the low freq modulation from the power or even fan. And the AD9910 pll most likely will unable to clean it.

marmeladapk, 2018-06-13:
_From @jordens on [2017-12-20 15:49](https://github.com/sinara-hw/sinara/issues/354#issuecomment-353100109)_

@a-shafir the reference is clean. And that hypothesis is already excluded be the other observations.

marmeladapk, 2018-06-13:
_From @jordens on [2018-01-02 19:00](https://github.com/sinara-hw/sinara/issues/354#issuecomment-354847897)_

AD9912 variant also works fine (https://github.com/m-labs/artiq/commit/a940550e4755f92c96408acfd560099e2e2ee3ed)
@gkasprow did you get a chance to hunt for the 1.2 kHz spurs?

[47 more replies]

closed 2019-09-02 by jordens