sinara-hw/Urukul #5: meta

hartytp, 2018-07-30:
Issues from sinara-hw/meta that could affect Urukul:
- https://github.com/sinara-hw/meta/issues/4 - mechanical issue. I'd also be happy just dropping the plastic washers on the default front panel now that the SMAs aren't floated by default
- https://github.com/sinara-hw/meta/issues/3
- https://github.com/sinara-hw/meta/issues/2
- https://github.com/sinara-hw/meta/issues/1

@gkasprow I think we should probably adopt the same FP grounding configuration for Sampler and at least add DNF resistors that can be used to tie the FP to circuit ground. Please can you confirm whether you think the new front panels will provide a solid grounding connection between the PCB and chassis ground (sorry if you've already said that elsewhere and I just missed it!).

gkasprow, 2018-07-30:
The new panels has special contact sprints that scratch the panel oxidization layer and touch the chassis rail. So providing that we use toothed lock washers, they will ensure contact with chassis. They also provide contact with neighbouring panel.

hartytp, 2018-07-30:
@jordens @sbourdeauducq in that case, here is what I propose. We'll get a batch of EEMs made with the new panels and the 0R resistors populated to short the panels to the circuit ground. We'll test them carefully to ensure that the ground contact between chassis and EEMs is solid and robust. We'll check them for EMI/ESD issues (we can also lend some to others for testing if that's helpful).

If that all looks good, then I suggest that we make the 0R resistors the default population option.

gkasprow, 2018-08-04:
changing to 0R then.
Added RC to front panel mounting holes.

gkasprow, 2018-08-04:
I swapped EEM connectors as well. It took a few hours of work but I believe it was worth the effort.

hartytp, 2018-08-04:
:+1: thanks for doing that. It's a lovely design

closed 2018-08-06 by gkasprow

hartytp, 2018-08-06:
@gkasprow are you happy that https://github.com/sinara-hw/meta/issues/4, https://github.com/sinara-hw/meta/issues/3 and https://github.com/sinara-hw/meta/issues/2 are also fixed?

gkasprow, 2018-08-06:
yes