sinara-hw/Urukul #8: Defloat the SMAs

hartytp, 2018-07-30:
@gkasprow didn't we agree to defloat the SMAs on Urukul by replacing the 100k resistors with 0R resistors (and adding a note that they can be floated by changing the resistors).

hartytp, 2018-07-30:
Note, I don't really mind what we do here, so if people prefer to keep them floated by default then that's fine by me. Just seems like unfloating them and scrapping the insulating washers might be easier and mechanically better. It's pretty easy for users who want the SMAs floated to drill out the FP and add the washers.

sbourdeauducq, 2018-07-31:
Drilling front panels is messy, better keep the washers in all cases.

hartytp, 2018-07-31:
> Drilling front panels is messy, better keep the washers in all cases.

Fine by me.

The things I haven't got a feeling for are:
1. how niche a use-case is floating the SMAs? For us, it will be sufficiently rare that I wouldn't mind having to modify/order new FPs to do it. We'd need user feedback here to know what other people think
2. What are the mechanical/cost downsides involved with the isolating washers. If they're cheap and mechanically fine then there isn't any point removing them. @gkasprow

gkasprow, 2018-08-04:
Cost of the washers is negligible. It's far easier to keep them and have same front panels.

closed 2018-08-04 by gkasprow