sinara-hw/Urukul #10: Depopulate screening clips?

hartytp, 2018-08-06:
I'm happy to leave them in, but removing them saves a BOM line.

We've tested Urukul quite extensively at this point, including in a rack with Kasli as an EMI source. We don't see any evidence of cross-talk or pickup at a level that bothers us (which shows that @gkasprow's routing is good!).

gkasprow, 2018-08-06:
we can make them DNP

hartytp, 2018-08-06:
any objections? @jordens

jordens, 2018-08-06:
OK. But I'd still like to see a measurement of the crosstalk (kasli to urukul, urukul to urukul, and channel to channel) with the cans mounted on one of the existing boards. The crosstalk channel channel to channel with the cans not mounted. was done in December.

hartytp, 2018-08-06:
"OK" as in "okay, let's remove them" or as in "okay in principle, but let's get some data first"?

Would be an interesting measurement to make, but we'd need to get some screening cans made first.

jordens, 2018-08-06:
DNP them now, get the cans made for an existing board, and measure. Same goes for Zotino and Sampler.

closed 2018-08-06 by gkasprow

klickverbot, 2018-08-06:
Ack. The one caveat I would add is that it can be quite difficult to provoke EMI problems without proper test equipment (although I guess the main worry for Urukul would just be picking up single RF tones at inconvenient frequencies, which we could simulate using a loop antenna and the Frankonia amp).

jordens, 2018-08-06:
Yes. But I think that's only difficult if you need to split the attenuation into emitted and immitted.
I didn't know Frankonia also builds amplifiers...