sinara-hw/Urukul #17: consistent LED sizes

jordens, 2018-11-13:
The front panel LEDs on Urukul are larger than all the others boards AFAICT. I am OK with Kasli being an exception in LED size, mounting and location (for space reasons). But Urukul should get LEDs similar in size to those on the other boards.

gkasprow, 2018-11-13:
We should definitely standardize aesthetics of the modules. So I fully agree.
The front panels are made for each revision anyway so there won't be any cost issues.

jordens, 2019-04-16:
I really like the edge emitting LEDs on the PCB bottom (Kasli, Humpback). That's an excellent use of space, they are surface mount, high density.

gkasprow, 2019-04-17:
The only drawback is aesthetics. LEDs are visible from a very narrow angle. There are miniature 1mm light pipes that can be inserted to the holes in the panel so the user experience would be much better.

jordens, 2019-04-17:
Yes. If those light pipes could be press fit into the panel that would be great.

gkasprow, 2019-08-16:
replaced with LED_DIALIGHT_571-0112F

closed 2019-08-16 by gkasprow