sinara-hw/Urukul #20: oscillation on P7V5, P7V0

jordens, 2018-12-18:
I am seeing an unexpected oscillation on Urukul-AD9910/v1.3 on both the P7V5 and P7V0 rails.
About 100mV peak-peak ~100 kHz, few overtones.
This isn't there on Urukul-AD9912/v1.0 and Urukul-AD9910/v1.0.
Any ideas @gkasprow

gkasprow, 2018-12-18:
can you check if it is before the L5? Does it have higher amplitude there?
It's quite possible that converter gets unstable, but It's surprisingly high frequency. DC/DC works a 1MHz.

gkasprow, 2018-12-18:
[image]

jordens, 2018-12-18:
IIRC I checked that side of L5 as well and there was no 100 kHz oscillation there. I would also expect that the LDO between P7V5 P7V0 attenuates the oscillation but it doesn't.

gkasprow, 2018-12-18:
It's also posible that power stage has some I=f(U) area that behaves like negative resistance and causes oscillation together with L5 and C183. Did you check outputs of IC5?

jordens, 2018-12-18:
Yes. As I mentioned there is oscillation at the RF output (after the choke), but less pronounced than on the LDO side of the choke.

gkasprow, 2018-12-18:
Are U sure these oscillations are not present at the input of the amplifier

jordens, 2018-12-18:
RF input? I didn't check that since this happens wihout any RF and upstream of the RF amplifier there isn't much else going on that I would suspect. But will check.

jordens, 2018-12-18:
I'll poke around tomorrow. I'll also check on another Urukul-AD9912/v1.1 (or v1.2) and another AD9910/v1.3.

jordens, 2018-12-19:
There seem to be a couple of major problems:

* The TPS62175DQC only supports 6V output max. We have 7.5 V
* The output voltage is typically around 7.25 V but changes with tolerances. Was that not measured when doing the initial testing last year?
* 250 mV is insufficient headroom for the 7V LDO which needs that at 150 mA (nominal current) but tolerances on the dividers shrink the available headroom.

gkasprow, 2018-12-19:
I remember that I measured that voltage and it was fine. But we work outside of specification so it may strongly depend on chip series.
Quick fix would be lowering the LDO voltage. It should not make big change to power stage if we operate it with 7 or 6.5V but big change to LDO.
If the LDO operates at the edge of its dropout voltage, it's not a big surprise to observe it oscillating.

gkasprow, 2018-12-19:
We use same DC/DC chip on Sampler, but there we have 6V so within spec.

jordens, 2018-12-19:
Even one of the first ones from last year (Urukul-AD9912/v1.0) has ~7.25 V on P7V5 at the switcher, ~6.95 V on P7V5, and a LDO dropout of 260 mV. Maybe the switcher has degraded over time. If the switcher circuit is the problem, then lowering the P7V0 might just mask it.

jordens, 2018-12-19:
"As a safety feature, the device clamps the output voltage at the VOS pin to typically 7.4 V, if the FB pin gets opened."

The proper fix is to stay well clear of that (e.g. 7 v) but also lower the p7v0 rail (to 6.5v).

jordens, 2018-12-19:
VOS is also 7 V Abs max Rating.

hartytp, 2018-12-19:
Good catches @jordens

jordens, 2019-01-07:
@gkasprow Some oscillation is still there even with the setpoints changed to P7V5=6.9V and P7V0=6.4V. It is only on P7V5A and P7V0A, not P7V5 and P5V0A. About 120 kHz, roughly sinusoidal but slightly asymmetric rise and fall, 80mV pk-pk. And It has opposite sign on P7V5A than on P7V0A (i.e. accross IC21). If I bypass L5 it becomes smaller on P7V5A (=P7V5 then) but doesn't change on P7V0A.
If I lower P7V0A even further it doesn't change. Ideas?

gkasprow, 2019-01-07:
try to get rid of C175 or lower it to 100pF

Datasheet says "Bypass  capacitors,  used  to  decouple  
individual  components  powered  by  the  LT1763-X,  will  
increase the effective output capacitor value. With larger 
capacitors  used  to  bypass  the  reference  (for  low  noise  
operation), larger values of output capacitors are needed. 
For 100pF of bypass capacitance, 4.7μF of output capaci
tor is recommended. With a 1000pF bypass capacitor or 
larger, a 6.8μF output capacitor is recommended"

jordens, 2019-01-08:
100 pF does not oscillate. 1 nF oscillates. From the 22µF Tantal + 5x100nF + 4x1µF ceramic output bypass I don't see how that can happen.

jordens, 2019-01-08:
But IME not more than the 3 Ohm the datasheet mentions.

gkasprow, 2019-01-08:
true.

hartytp, 2019-01-08:
@gkasprow do you know which part numbers TechnoSystem used for the decoupling capacitors?

hartytp, 2019-01-08:
I can believe that cheap 1uF ceramics could have quite low capacitance when biased to 7V. If the tantalum they used has a high ESR/ESL as well then that would explain this.

hartytp, 2019-01-08:
@gkasprow the TPSB226K010R0700 capacitor you've used as an example for the 22uF decoupling capacitor is only 10V, with a category voltage of 7V. That seems a bit marginal, doesn't it? Won't be good for board lifetime.

jordens, 2019-01-08:
That's still only about 1e-6 failures per 1e3 hours at 50C.

jordens, 2019-01-08:
And the Tantalum cap has 0.7 Ohm ESR.

hartytp, 2019-01-08:
> And the Tantalum cap has 0.7 Ohm ESR.

It's generic in the BOM isn't it? The part above is only an example, AFAICT they are free to substitute it if they wish.

hartytp, 2019-01-08:
> That's still only about 1e-6 failures per 1e3 hours at 50C.

The references I'd seen suggested it would be somewhat worse than that, but yes, the failure rate will still be low. My comment wasn't meant to imply that this is the cause of the oscillations, just that it's a bit more marginal than I would have liked.

hartytp, 2019-01-13:
My best guess for this is that it's the 3x1uF ceramics causing oscillations. If we ignore the 22uF tantalum (ESR/ESL too high compared with the ceramics so doesn't contribute) then we would have 3uF total capacitance with very low ESR. From the data sheet, that's clearly in the unstable regime.

@jordens can you try adding a 4u7 ceramic near the LDO output and see if that stabilizes the LDO. Otherwise, @gkasprow is it worth simulating the circuit in LTSpice to check for stability?

jordens, 2019-01-13:
It's 4.5 µF ceramics and the Tantalum needs to have high ESR (> 3 Ohm) to not contribute. Not clear at all.
I have modified all affected boards already.

hartytp, 2019-01-13:
ack. @gkasprow can you ask a student to look at this? We should aim to fix this issue before producing any more boards. AFAICT Mirny uses the same supplies so will have the same bug unless we track it down.

gkasprow, 2019-01-13:
I sketched such circuit 
[image]

gkasprow, 2019-01-13:
And I do not see big difference whether Cbyp is 100pF or 10nF
With C=100p
[image]
With 10nF
[image]

gkasprow, 2019-01-13:
2u cap has 1mOhm ESR, 10u cap ha 0.7 Ohm ESR. Vin = 7.4V Load pulse is 100mA

klickverbot, 2019-01-13:
> all affected boards 

Are these all your 1.3 boards, or only a subset?

jordens, 2019-01-14:
Every board I looked at.

gkasprow, 2019-01-27:
I plan to replace the TPS62175 with [NCP3170](https://www.onsemi.com/pub/Collateral/NCP3170-D.PDF) which already use in many designs. Another option is [LMR36015](http://www.ti.com/product/LMR36015).
The main requirement is continuous operation (no sleep or pulse skipping mode) and PG output.

hartytp, 2019-01-27:
@gkasprow will you have a look at the LDO output at the same time? My guess is that a 22uF ceramic will fix the oscillation (or a FB to decouple the LDO from the ceramics).

If we're going to produce a new hw revision then it's worth looking at the other outstanding issues at the same time, particularly https://github.com/sinara-hw/Urukul/issues/16

gkasprow, 2019-01-27:
We observe similar issues on AFCZ board with Exar converter. After some time one of the supplies switches off on all boards.

hartytp, 2019-01-28:
Ouch! So...how to we make sure this does not happen on Sayma 2? Scrap the exar? Better caps? More careful loop tuning? Whatever we do I don’t want to have to deal with converter instabilities any more :)

sbourdeauducq, 2019-01-28:
@gkasprow @hartytp I suppose you want to discuss this in https://github.com/sinara-hw/sinara/issues/567

hartytp, 2019-01-28:
@gkasprow Once this is fixed we should also add a note to the wiki with a change list that users can apply to their boards to fix/work around this issue.

gkasprow, 2019-08-16:
[image]

hartytp, 2019-08-16:
NB looking again at the current Urukul design, there are other issues around IC22. E.g. the abs max rating for the sleep pin is 7V, however it's driven from the 7V5 output. As there is no current limiting resistor that's probably destructive.

hartytp, 2019-08-16:
@gkasprow the regulator PN recommends an effective output capacitance of 47uF with FSEL low. I'd guess that with a 7V5 bias that 22uF capacitor will be a bit on the small side, can we beef it up please?

hartytp, 2019-08-16:
Other than that, assuming all capacitors have sensible voltage ratings, LGTM.

hartytp, 2019-08-16:
Actually, one other thing: I'm still not totally comfortable with the situation re the LT1763. It seems odd to me that we're saying that we can't get it to work with the data sheet recommendation of 10nF bypass capacitor. My guess is that fiddling with that capacitor is masking some other problem

gkasprow, 2019-08-16:
@hartytp  I replaced IC22 with TPS62148 which works up to 12V at its output. True, the DS says we should use 47uF so I added another 22uF cap.

hartytp, 2019-08-27:
> Actually, one other thing: I'm still not totally comfortable with the situation re the LT1763. It seems odd to me that we're saying that we can't get it to work with the data sheet recommendation of 10nF bypass capacitor. My guess is that fiddling with that capacitor is masking some other problem

@gkasprow any thoughts about this?

hartytp, 2019-08-27:
Should we just switch IC21 to a different regulator that's a bit less fussy? Any reason not to just go for the same LT3045 that we use everywhere else?

gkasprow, 2019-08-29:
Replaced IC21 with LT3045.
[image]

closed 2019-08-29 by gkasprow