sinara-hw/Urukul #24: SMA stackup wrong, SMA cables loose

Jochenwolf3, 2019-04-08:
Using standard SMA cables (e.g. JP1XJP1XF316-40 from Gigatronix), there is not enough exposed threading to put the SMA cables on. The plastic washer, tooth washer, and nut take up so much space on the thread that upon tightening the SMA cable nut, the cable still feels loose and can be wiggled around, indicating an unreliable connection:

[image]

An extra ~3 mm of SMA threading is needed.

Workaround: remove the plastic washer from the stack, which creates just about enough space (and is fine for people who do not want their SMAs floated anyway)

Jochenwolf3, 2019-04-08:
This applies to Urukuls manufactured by both Creotech and Technosystem.

gkasprow, 2019-07-01:
One option is to switch to longer SMA connectors like Telegartner J01151A0921. But they are very long.
They are also far more expensive (30$/pc). 
 I can also move the PCB edge to the panel by 1mm. This should solve the issue.

gkasprow, 2019-08-16:
I moved them by 1mm to the panel.

closed 2019-08-16 by gkasprow