sinara-hw/Urukul #29: Test data for Urukul

TheCakeIsAPi, 2019-07-04:
I took some detailed test data to measure power levels and harmonics.  I think it may be useful to others.  I suggest creating a place on the Wiki to catalog this and other such data for each piece sinara-hw.

---

Readings for urukul0_ch0, AD9910 model, Urukul v1.3, s/n TS 92/19 0027, with 125 MHz external clock from Wenzel 501-31911, via Mini-Circuits ZB4PD-52-20W-S+ power splitter, measured on Keysight N9343C (which only has spec'd response from 1 MHz to 13.6 GHz).  Attenuation and reference level of the N9343C were set to minimum levels that support each measurement.  All readings were allowed to average until stabilized to +/-0.05 dBm 

At attenuation = 0.0 dB 

Setpoint [MHz] | Fundamental [dBm] | 1st [dBm] | 2nd [dBm] | 3rd [dBm] | 4th [dBm] | 5th [dBm] | 6th [dBm] | 7th [dBm] | 8th [dBm]
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0.1 | -21.14 | -59.03 | -54.93 | -93.07 | -73.38 | -94.07 | -84.78 | -91.77 | -96.61
0.5 | 4.51 | -15.45 | -11.61 | -25.02 | -24.35 | -51.70 | -35.14 | -34.46 | -37.85
1 | 7.67 | -16.80 | -12.32 | -18.27 | -29.25 | -30.87 | -34.51 | -39.28 | -39.84
10 | 10.67 | -12.69 | -13.94 | -26.12 | -27.76 | -36.11 | -55.32 | -43.85 | -42.65
20 | 10.86 | -24.90 | -13.65 | -22.87 | -28.67 | -47.68 | -35.85 | -35.45 | -38.48
50 | 10.74 | -14.18 | -15.01 | -27.57 | -29.01 | -38.05 | -51.52 | -44.53 | -42.71
100 | 9.70 | -33.59 | -16.72 | -34.36 | -26.81 | -40.14 | -41.07 | -43.88 | -56.89
200 | 8.97 | -22.22 | -16.23 | -24.89 | -30.49 | -37.97 | -37.79 | -38.80 | -40.14
300 | 8.27 | -19.17 | -19.51 | -29.80 | -34.75 | -38.90 | -51.92 | -53.38 | -57.95
400 | 7.68 | -17.82 | -21.60 | -33.04 | -37.80 | -50.37 | -57.45 | -59.80 | -64.68
500 | -1.80 | -41.57 | -51.71 | -72.36 | -89.35 | -91.63 | -93.15 | -84.54 | -107.57

At attenuation = 10.0 dB 

Setpoint [MHz] | Fundamental [dBm] | 1st [dBm] | 2nd [dBm] | 3rd [dBm] | 4th [dBm] | 5th [dBm] | 6th [dBm] | 7th [dBm] | 8th [dBm]
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0.1 | -27.06 | -81.35 | -62.09 | -97.37 | -84.11 | -103.78 | -91.37 | -100.48 | -104.22
0.5 | -3.2 | -37.82 | -52.21 | -66.76 | -77.86 | -85.92 | -86.37 | -97.59 | -120.76
1 | -0.43 | -34.47 | -47.80 | -75.28 | -86.45 | -101.91 | -93.22 | -96.14 | -106.71
10 | 1.95 | -31.04 | -28.23 | -51.76 | -57.29 | -76.26 | -78.15 | -83.85 | -80.20
20 | 2.10 | -33.05 | -28.30 | -54.50 | -52.31 | -72.39 | -70.96 | -82.98 | -82.58
50 | 1.89 | -33.24 | -28.50 | -52.67 | -48.35 | -74.77 | -77.26 | -79.33 | -73.58
100 | 0.80 | -38.51 | -63.22 | -61.73 | -71.97 | -97.45 | -97.67 | -107.40 | -93.03
200 | 0.05 | -38.25 | -42.16 | -63.01 | -84.55 | -82.66 | -108.85 | -116.62 | -99.45
300 | -0.51 | -35.91 | -48.83 | -82.43 | -100.53 | -111.79 | -118.62 | -120.05 | -97.72
400 | -1.20 | -38.37 | -49.77 | -89.45 | -74.66 | -108.12 | -116.75 | -114.08 | -102.29
500 | -11.20 | -61.47 | -77.59 | -74.73 | -100.23 | -93.12 | -99.83 | -86.71 | -112.63

At attenuation = 20.0 dB 

Setpoint [MHz] | Fundamental [dBm] | 1st [dBm] | 2nd [dBm] | 3rd [dBm] | 4th [dBm] | 5th [dBm] | 6th [dBm] | 7th [dBm] | 8th [dBm]
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0.1 | -31.06 | -82.29 | -68.34 | -109.04 | -92.48 | -111.23 | -99.94 | -109.85 | -112.36
0.5 | -11.99 | -56.69 | -71.73 | -95.76 | -101.86 | -114.37 | -102.81 | -106.94 | -116.72
1 | -9.94 | -54.54 | -56.49 | -89.12 | -105.94 | -110.93 | -102.79 | -107.01 | -117.29
10 | -7.89 | -50.19 | -57.35 | -91.36 | -97.88 | -107.95 | -103.53 | -96.04 | -108.26
20 | -7.79 | -52.72 | -58.03 | -90.75 | -99.82 | -102.07 | -101.55 | -104.73 | -103.31
50 | -7.96 | -52.36 | -59.26 | -84.44 | -87.55 | -86.88 | -97.76 | -92.61 | -83.19
100 | -9.04 | -57.40 | -61.76 | -78.50 | -91.80 | -117.64 | -107.40 | -112.64 | -102.07
200 | -9.73 | -57.39 | -72.31 | -72.66 | -93.26 | -95.95 | -125.22 | -122.35 | -130.24
300 | -10.27 | -58.65 | -74.60 | -109.24 | -107.74 | -115.75 | -125.36 | -124.54 | -98.86
400 | -10.94 | -59.62 | -79.36 | -98.48 | -74.72 | -111.95 | -119.18 | -114.63 | -104.34
500 | -21.00 | -78.52 | -99.07 | -74.91 | -99.55 | -92.91 | -103.02 | -87.33 | -114.87

At attenuation = 31.5 dB 

Setpoint [MHz] | Fundamental [dBm] | 1st [dBm] | 2nd [dBm] | 3rd [dBm] | 4th [dBm] | 5th [dBm] | 6th [dBm] | 7th [dBm] | 8th [dBm]
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0.1 | -37.89 | -85.04 | -77.41 | -122.04 | -114.29 | -115.58 | -110.65 | -120.06 | -123.70
0.5 | -22.38 | -71.24 | -89.84 | -107.81 | -108.76 | -127.83 | -114.12 | -118.34 | -127.07
1 | -21.01 | -72.10 | -90.08 | -111.97 | -111.30 | -127.43 | -114.38 | -118.07 | -128.06
10 | -19.22 | -72.13 | -90.74 | -110.14 | -105.28 | -114.04 | -113.51 | -94.85 | -116.15
20 | -19.28 | -75.95 | -94.72 | -91.71 | -107.55 | -112.85 | -112.24 | -116.33 | -114.02
50 | -19.27 | -74.93 | -92.21 | -95.77 | -101.06 | -97.92 | -108.30 | -103.60 | -93.96
100 | -20.27 | -79.05 | -87.48 | -89.73 | -104.00 | -117.98 | -112.12 | -110.51 | -105.80
200 | -21.19 | -78.33 | -106.81 | -82.70 | -92.31 | -109.93 | -133.86 | -120.94 | -102.95
300 | -21.58 | -80.96 | -112.44 | -110.40 | -108.11 | -115.68 | -122.51 | -125.25 | -99.63
400 | -22.44 | -82.73 | -105.55 | -98.03 | -74.84 | -113.93 | -119.41 | -114.93 | -104.55
500 | -31.73 | -93.37 | -99.74 | -75.03 | -99.27 | -92.84 | -104.14 | -87.46 | -116.22

jordens, 2019-07-04:
With "1st" you mean the first overtone but the second harmonic, right?

TheCakeIsAPi, 2019-07-05:
@jordens yes.  Fundamental = setpoint.  1st = 2 x setpoint.  2nd = 3 x setpoint.  Sorry if that was unclear.  I did also see some other peaks that were on occasion bigger than the nearby harmonics, but I couldn't think of a decent way to tabularize that, so those are not recorded here.

dhslichter, 2019-07-05:
Thanks for the data, @TheCakeIsAPi!

> I did also see some other peaks that were on occasion bigger than the nearby harmonics, but I couldn't think of a decent way to tabularize that, so those are not recorded here.

One way would be to have a column for "largest spur", so that people can see the spur-free dynamic range.  Another thing that might be nice (for a final table in a wiki) would be to convert the harmonics into dBc (relative to fundamental), to help save on the mental gymnastics of the doing the math each time.

jordens, 2019-07-24:
Some of the the data look suspicious to me. E.g.

* A -26.4 dBc third harmonic at 100 MHz, 0 dB is much worse than what I measured way back (AD9912 v1.0, -53.5 dBc third harmonic at 80 MHz, 0 dB).
* At 10 dB attenuation for 10, 20, 50 MHz the third harmonic is 20-35 dB worse than for both 1 or 100 MHz.
* At 0 dB attenuation the high harmonics of 400 MHz (2.8, 3.2, 3.6 GHz) at -57, -60, -65 dBm (w.r.t. 8 dBm fundamental).
* Many measurements appear to be the dynamic range/floor of the spectrum analyzer and not the power of the harmonics.

> Attenuation and reference level of the N9343C were set to minimum levels that support each measurement

Minimizing input attenuation (below the typical/automatic `min(0 dBm, ref_level) + 10 dB`) in spectrum analyzers is risky. Sure you didn't create harmonics yourself?

closed 2019-08-27 by hartytp