sinara-hw/Urukul #30: redo CPLD wiring

jordens, 2019-07-15:
Redo the CPLD wiring to avoid pitfalls, decouple the channels, remove hidden state, and improve interoperability with the different software modes.

* PROFILE should be per-channel (+9)
* Attenuator access should be per-channel (CS/LE) (+3 for CS/LE, +3 for MISO)
* OSK is useful if per-channel (+4) c.f. #19 
* add hardware revision resistors (+2)

Compensate pin usage (+21) with:

* Share Master_RESET and DDS_RESET (-1)
* Use shared MOSI/SCK for DDS (-6)
* shrink IFC_MODE (let's say -2)
* fewer test points (up to -3)
* bind OSC_ENn to MMCX_OSC_SEL (see Mirny) (-1)
* fix the configurable EEM LVDS direction RE_DE (up to -16)
* fix FSEN (-1)
* DDS_SYNC_CLK0 (-1)
* bind green LED to RF switch (-4)

tballance, 2019-08-07:
+1 for pinning out the profile pins. This would be really useful for us.
When the CPLD code is being revisited, I think it would also be useful to add more detailed version and build info which can be read out through the SPI interface.

gkasprow, 2019-08-08:
we can also use a bigger BGA256 package and route all pins without any compromises. 
Don't we want more CPLD resources?

jordens, 2019-08-08:
Right. A small fpga with lvds would make the 4 lvds cmos converters redundant as well. Might be worth it.

gkasprow, 2019-08-08:
FPGAs have usually much higher pin to pin delays than CPLDs and higher jitter. And it is much more complex modification than just replacement of the CPLD.

jordens, 2019-08-08:
Sure. Just pointing out the option.
A BGA CPLD with a bit more resources would be nice to alleviate the pin and resource constraints.

tballance, 2019-08-08:
From a control perspective, it seems to me that it would be nice to expand the CPLD shift register to include some bits for addressing. This would allow us to read out more detailed version info (like a human-readable build string) and also expand the ability to operate in different modes. 
A mode that could be useful for us would allow us to control the profile pin switches from a fifo in the CPLD which is programmed through SPI writes and clocked out with a strobe pulse from the EEM interface. This would enable us to make bursts of fast switches (<1us) between profiles which would not be achievable through SPI writes alone.
Also nice would be to have the CPLD interface backwards compatible, in that it boots in 'legacy mode' and can be switched (through a write sequence) to the expanded mode if the proto_rev version is above a threshold. From a user perspective I think this would be nice, as many people will have mixed hardware and making it 'plug-and-play' will hopefully reduce the number of issues about hardware not working.

gkasprow, 2019-08-16:
@jordens do you think adding I2C to SPI -> JTAG converter would be useful as it is in case of the Banker?

jordens, 2019-08-16:
To jtag the cpld over i2c? Probably not that high value in practice. And part of the tooling is missing.

gkasprow, 2019-08-18:
@jordens please review the CPLD signal assignments
https://github.com/sinara-hw/Urukul/releases/tag/v1.4rc1

jordens, 2019-08-19:
I'm on vacation until the end of the week. Will review then.

jordens, 2019-08-26:
@tballance These are all nice ideas. Some of them is (like the multiple registers and the addressable SPI switch) is planned/implemented in Mirny and looking for funding (https://github.com/quartiq/mirny/issues/1). The same would apply to these features and the gateware/software for Urukul v1.4.

jordens, 2019-08-26:
@gkasprow I looked at the schematic and ported the current gateware.

* Could you split the FSEN signal between the EEM0 and EEM1 port? Since it introduces an imbalance, and since we only ever support leaving EEM1 open, it makes sense to treat them differently and force EEM0 to be always non-fail-safe but with more symmetric thresholds while EEM1 can be fail safe but asymmetric. NB: This doesn't affect the SYNC duty cycle asymmetry as that signal doesn't go through those converters.
* `Profile_P1` `ATT_S_IN`, `ATT_S_OUT`, `ATT_LE` don't exist anymore
* Looks good otherwise.

gkasprow, 2019-08-29:
done

closed 2019-08-29 by gkasprow