sinara-hw/Urukul #31: v1.4 changelog

gkasprow, 2019-08-16:
- changed TPS62175DQC to TPS62148RGX (#20)
- C175 10nF ->100pF (#20 )

hartytp, 2019-08-16:
I did a basic design review of all ICs on Urukul focussing on:
- abs max ratings
- typical application circuits
- recommended capacitors
- skim of important device parameters like drop out voltages

didn't see anything other than the reported issues. NB didn't look at overall design or info not on the schematic (capacitor voltage ratings etc)

hartytp, 2019-08-18:
Great, thanks for doing all that! @gkasprow can you confirm that all open issues marked with a v1.4 milestone are now complete (I just did some tagging, so please re-check). If so, great!

hartytp, 2019-08-27:
@gkasprow from a quick look, it seems that a few of these issues still aren't fixed. I'm keen to get v1.4 off to manufacture asap. Can you have a go at fixing them and then upload a new rc? Thanks!

hartytp, 2019-08-29:
@gkasprow ping...what's the timeline for the next Urukul revision

gkasprow, 2019-08-29:
I'm still out of the office but I'm slowly working on it.
I will be back in 4 days.

hartytp, 2019-08-29:
That's fine, we can wait a few more days (enjoy your holiday!). Thanks for the update.

gkasprow, 2019-08-29:
done.

hartytp, 2019-08-29:
> added CVHD950 oscillator Vctrl input pin divider. It may be needed when VCXO oscillator is installed in case of poor availability of the standard part. (#18)

Remind me why we'd ever want a VCXO and not an XO?

hartytp, 2019-08-29:
I've reviewed the schematics. Other than #33 I don't see any issues.

Once you've fixed that and @jordens has signed off, let's send these cards to order (we have a Creotech order that's been blocked by this release for quite a while).

gkasprow, 2019-08-29:
We don't need VCXO. It is just in case we use VCXO due to poor availability of XO

hartytp, 2019-08-29:
Ok

hartytp, 2019-08-29:
Oh and can you add the dds pn to the panel? Then I think we’re good to go

hartytp, 2019-09-01:
Thanks @gkasprow!

Assuming @jordens is happy to sign off, let’s send this off to manufacture

gkasprow, 2019-09-01:
I'm just checking with PI simulations if everything is fine.

gkasprow, 2019-09-01:
It looks like I didn't break anything related to power delivery network.

gkasprow, 2019-09-01:
I drafted the new [release](https://github.com/sinara-hw/Urukul/releases/tag/v1.4rc3). Please have a look at the panels.

hartytp, 2019-09-02:
ping @jordens

jordens, 2019-09-02:
I'm ok with the changes as described and did a test-port of the CPLD gateware. But I didn't review anything else.

hartytp, 2019-09-02:
Okay, I've reviewed everything apart from the CPLD changes. So, if you've checked that then I'm happy to cut this release here.

Thanks all!

closed 2019-09-02 by hartytp