sinara-hw/Urukul #32: Ic18 drop out

hartytp, 2019-08-16:
Are you definitely happy with the drop voltage for IC18? We don't quite hit the recommended 3.25V for Vbias-Vo (particularly when tolerances are accounted for) so it seems a bit marginal to me from the data sheet...

hartytp, 2019-08-16:
Well, maybe it's okay at the currents we're operating at?

gkasprow, 2019-08-16:
We can always switch to 7V5

gkasprow, 2019-08-16:
ah, max bias is 5.5V

gkasprow, 2019-08-16:
We can also slighty increase the 2V rail.

gkasprow, 2019-08-29:
I increased the P2V0 rail to P2V1 to satisfy Vbias requirements.

closed 2019-08-29 by gkasprow

hartytp, 2019-08-29:
Sounds good. The small change in power efficiency is worth it to make sure the LDO functions as expected.

Did you make sure to update the power budget table?

gkasprow, 2019-08-29:
done
[image]