sinara-hw/Urukul #34: Fixing Urukul v1.3, v1.2 defects

jbqubit, 2019-08-30:
What's the prospect for applying patches to existing Urukul v1.3 to fix the most important problems addressed in the v1.4 roll? There's many dozens of these boards in circulation. It would be nice to have a repair path that addresses the most critical bugs without replacing the boards. 

From https://github.com/sinara-hw/Urukul/issues/31 v1.3 -> v1.4 the fixes are
- replace TPS62175DQC... is there a pin compatible replacement?
- C175 10nF ->100pF... easy patch
- pulled OSK to P3V3_DVDD_IO... easy patch
- R87 : 1k8 -> 160R... easy patch
- changed IC21 to LT3045... is there a pin compatible replacement? 

I don't have a clear picture of the v1.2 -> v1.3 mods. What are the prospects for similar patching?

hartytp, 2019-08-30:
@jbqubit firstly, I'm going to close this since it's not really suitable for an "issue".

> What's the prospect for applying patches to existing Urukul v1.3 to fix the most important problems addressed in the v1.4 roll? There's many dozens of these boards in circulation. It would be nice to have a repair path that addresses the most critical bugs without replacing the boards.

The short answer is that this is very doable, but someone needs to organise it. Probably the easiest path is to email TechnoSystem/Creotech and discuss it with them. FWIW I don't have any plans to patch our boards any time soon (we don't have enough spare hardware to swap them out) and I'm not aware of anyone else planning to do it.

> From #31 v1.3 -> v1.4 the fixes are

I think that's a complete list, but if you want to be sure then you'll need to do some archaeology on the issue tracker I'm afraid (it's easy for changes to be left off the change list).

> I don't have a clear picture of the v1.2 -> v1.3 mods. What are the prospects for similar patching?

Again, without stating the obvious, the prospects are that this can happen if you or someone else put the time into organizing it. You'd need to go through the issues and figure out what the changes were, then figure out which if any patches were applied to your boards by hand after production (IIRC this wasn't really documented anywhere). Then contact TS or whoever you want to do the rework and arrange for it to be done.

This would all be easier if there were a bit more discipline about the work-flow in Sinara...

> replace TPS62175DQC... is there a pin compatible replacement?

The part itself needs to be replaced as it's been operated outside its abs max ratings and is most likely damaged (e.g. the PG transistor is probably gone so the sequencing won't work correctly which may or may not actually matter in practice). However, it does not necessarily need to be replaced with a different model -- IIRC you may be able to get away with just changing the set voltages on some of the rails See [here](https://github.com/sinara-hw/Urukul/issues/20). However, whoever is organizing this would need to think this fix through and check it's okay.

> C175 10nF ->100pF... easy patch
> changed IC21 to LT3045... is there a pin compatible replacement?

IIRC Changing the cap should be enough without swapping the LDO for a different PN. Again, see the discussion on the issue.

> pulled OSK to P3V3_DVDD_IO... easy patch

Not necessary unless you want to use that functionality. See the issue.

> R87 : 1k8 -> 160R... easy patch

I don't think this is strictly necessary, but sure, probably worth doing.

closed 2019-08-30 by hartytp

gkasprow, 2019-08-30:
v1.2 -> v1.3 main change was the addition of the clock mux.
But one needs to fix the oscillations by changing the RC values.
DC/DC converter and LDO can be left as they are. Even though the buck converter operates slightly outside its spec, it does not cause any problems. ICs are usually designed with some margin.
The same with R87 - if your board doesn't have a problem with 1V8 rail, don't change the resistor.

hartytp, 2019-08-30:
> v1.2 -> v1.3 main change was the addition of the clock mux.

Okay, well that's not required.

> DC/DC converter and LDO can be left as they are. Even though the buck converter operates slightly outside its spec, it does not cause any problems. ICs are usually designed with some margin.

IIRC the PG line was quite far outside its max ratings, and may well be damaged so I wouldn't be surprised if the power sequencing for that chip doesn't work. Having said that, small sequencing issues probably won't cause any harm.

gkasprow, 2019-08-30:
The absolute maximum rating for PG and VOS is 7V. We have 1k2 pullup to 7.5V. and IC22 VOS is 7.5V in our case.
So, the reduction of the IC22 output voltage to 7V and reduction of the IC21 voltage to let's say 6.5V solves the issue. The power stage will get a slightly lower voltage which will limit its max output power.

gkasprow, 2019-08-30:
Altium has a nice feature of document comparison.
Open v1.2 and 1.3 projects, then  Project -> show differences, then advanced mode, select an old version in the left panel, new version in the right panel and you will get a list with changes.
This function compares logical changes - it ignores cosmetics.
There is another function Project -> show physical differences that detects all changes, even cosmetical. 
It must be done for every schematic page.
I compared v1.2 with v1.3 in this way. The only change was made in CLK_INPUT and Urukul schematic sheets:
[image]
[image]

gkasprow, 2019-08-30:
So, v1.2->v1.3 also added 0R resistors to ground floating connectors and front panel grounding.

hartytp, 2019-08-30:
> The absolute maximum rating for PG and VOS is 7V. We have 1k2 pullup to 7.5V. and IC22 VOS is 7.5V in our case.

Wasn't there also an issue with the max current being violated? Although, maybe that's fine as only a brief transient?

hartytp, 2019-08-30:
> Altium has a nice feature of document comparison.

I didn't know about that. That's a nice feature

jbqubit, 2019-09-03:
Thanks for the thorough replies! As suggested by @hartytp we can also add this to Wiki so users know what patches they might apply themselves (or send back to manufacturer for repair). 

XXX == @gkasprow please give explicit list of component changes so that that Technosystem/Creotech can develop workflow. 

# patches v1.3 -> v1.4

- TPS62175DQC
    - replace TPS62175DQC as it may be damaged
    - changes to prevent over voltage again XXX

- C175 : 10nF ->100pF

- R87 : 1k8 -> 160R

- IC22 : reduce output to 7.0 V -- XXX

- IC21 : reduce output to 6.5V --  XXX

# patches v1.2 -> v1.3 
- Skip modifications related to clock mux. A patched v1.2 board should logically behave like a v1.2 board. 

- add 0R resistors to ground floating connectors -- XXX

- add 0R resistors to front panel grounding -- XXX

hartytp, 2019-09-03:
> As suggested by @hartytp we can also add this to Wiki 

? I didn't suggest that, but sure, go for it.

jbqubit, 2019-09-03:
> ? I didn't suggest that, but sure, go for it.

Github remembers the ghost of @hartytp past. :) 

https://github.com/sinara-hw/Urukul/issues/20#issuecomment-458237420

hartytp, 2019-09-03:
fair enough. my memory for these things doesn't stretch back that far.

gkasprow, 2019-09-03:
@jbqubit U need to modify 2 additional resistors. I updated your post above

jbqubit, 2019-09-03:
Thanks @gkasprow! 

>add 0R resistors to front panel grounding -- no pads foreseen for such option.

So a strap from accessible board ground to the front panel?

gkasprow, 2019-09-03:
Yes, you can do such a strap with a piece of wire.

jbqubit, 2019-09-03:
Thanks everyone. I've created an [Errata](https://github.com/sinara-hw/Urukul/wiki#errata) section of the wiki calling out these patches.