sinara-hw/Urukul #37: phase lock application

jordens, 2019-11-27:
Follow what transpires in https://github.com/sinara-hw/mirny/issues/10
(MMCX for VCXO tuning and jumper to choose between the first stage mux and the second stage mux, even though that PFD frequency only works for AD9912)

hartytp, 2019-12-31:
At a minimum, (space permitting) I think we should add a DNPd MMCX along with a pads for an RCRC filter/attenuator to allow external tuning of the VCXO. Probably also best to add a diode to protect from negative voltages (e.g. I want to use the +-10V stabilizer DAC to tune the VCO).

However, I'm not sure how useful that will be in practice since ground loops will likely lead to lots of spurs on the output. Low frequency noise can be rejected by the feedback, but anything at kHz will make it through largely unattenuated in most applications.

So, again layout/space permitting, what about adding a unity-gain inamp here? Since we don't have a negative rail we'll make it single supply. Copy the EMI filter/protection (minus the CMC) from Sampler and choose a fixed-gain low noise InAmp that can run rail-rail from a single supply. Then have an attenuator + RC filter afterwards to match 0->10V into the VCXO with limited noise BW.

If we had a negative rail we could consider making it work with the full +-10V stabilizer output range, but we don't and it's not worth adding one just for this.

@gkasprow what do you think?

gkasprow, 2019-12-31:
Some OAs (i.e. LM358) work with inputs below GND, the gain would need to be <0.1 so it should be possible with a single supply. The oscillator tuning is usually referenced to 1/2 Vcc which makes it even simpler for diff amplifier configuration.

hartytp, 2019-12-31:
Sounds good. Let's go for something like that then.

gkasprow, 2020-02-19:
what about such circuit?
[image]

gkasprow, 2020-02-19:
version with opamp bypass
[image]

gkasprow, 2020-02-19:
let's make the opamp and MCX DNP by default

hartytp, 2020-02-19:
@gkasprow what about the flexible clock routing (0R jumpers to select whether each channel is driven by the output of the first or second mux?) is it too late to sneak that in?

hartytp, 2020-02-19:
@gkasprow as a basic circuit that looks like it should work (I’ll double check the resistor values later). I’d expect poor cmrr from that diff amp. Iirc there are some InAmps that can take inputs beyond their supply range which would do the job beautifully don’t have a pn to hand right now. A resistor ladder to make a precision differential input divider would also do the job I guess.

Nit picks:
- we could consider splitting the 10k resistor that biases the circuit from the 3V3 to make an RCR low pass filter.
- We could also put a small input cap to limit the bw/filter EMI

Both of those are low priorities. Would be nice to do something to improve the CMMR if we can but the current implementation should still be useful as is.

marmeladapk, 2020-02-19:
@hartytp We're sending this project to manufacturer right now, without including your nit picks since you said that they're low priority. (and right now we have the last chance to swap production files).

jordens, 2020-02-19:
@marmeladapk I [pointed out specifically](https://github.com/sinara-hw/Urukul/issues/45#issuecomment-587699998) yesterday that the clocking jumpers should not be omitted...

hartytp, 2020-02-24:
Closing and breaking out into two separate issues for clarity...

closed 2020-02-24 by hartytp