sinara-hw/Urukul #39: pre-amp

hartytp, 2019-12-03:
Consider having a design variant (or just changing the main design) with a lower-gain pre-amp.

I've found that the 23dB gain provided by the ERA3-SM+ can be annoyingly high as a default option. I suspect that we can find something footprint compatible with lower gain/P1dB and lower current draw.

Thoughts?

dnadlinger, 2019-12-03:
As per the other issue, MAR1-SM+ is a potential pin-compatible alternative, but there are probably better options.

dnadlinger, 2019-12-03:
See https://github.com/sinara-hw/Booster/issues/335#issuecomment-561401294 for context.

hartytp, 2019-12-03:
hmm...no...I'd forgotten there's already a 9dB attenuator on Urukul. So the gain is actually more like 14dB.

An ADL5535 would probably have been a better option. We could have scrapped the 7V/7V5 rail completely and only had a 5V rail. Gain stability/tolerance looks better, power consumption is lower.

dnadlinger, 2019-12-03:
Yep, see https://github.com/sinara-hw/Booster/issues/335 :)

ADL5535 has quite a bit more current draw, but, as you point out, from 5V which we have elsewhere.

hartytp, 2019-12-03:
That's true, I'd mis-remembered the MAR1-SM+ current draw.

I think the general statement from me is that Urukul has ~10dB higher gain than I'd ideally like, even with the 9dB attenuator (which seems a little silly from a NF perspective, but I guess the DDS noise still dominates).

gkasprow, 2019-12-03:
It's not footprint compatible. What about lowering the operating current by increasing the resistor in an existing design? Gain should also drop

hartytp, 2019-12-03:
Looking again at our use-case, the ideal gain for us is somewhere between 0dB-5dB. We're currently at 14dB with the attenuator/amp combo. 

So, maybe a pre-amp bypass is the best option here. Could either be a population option or just made the default. If we do scrap the pre-amp then we can move the 7V5 rail to 5V5 and gain small efficiencies elsewhere at the same time.

Anyway, just something to think about for next time we put out a release...

jordens, 2019-12-04:
Up to 10 dB less gain is ok with me. But do look at the distortion/ip3. I don't want to fight with harmonics at 0 dBm output. And would like to be able to easily reach that at all frequencies.

jordens, 2019-12-04:
But why did you design booster to have so much gain? Surely you would have designed Booster to match the Urukul output range. And you wouldn't change Urukul just because Booster has so much gain.

dnadlinger, 2019-12-04:
> And you wouldn't change Urukul just because Booster has so much gain.

Agreed on this – the beefy Urukul output is mildly convenient sometimes.

hartytp, 2019-12-04:
Nb that phaser doesn’t have the pre-amp so booster needs the gain to work with it.

hartytp, 2019-12-04:
@dnadlinger when have you found the urukul pre-amp convenient? I’ve only ever found it a bit annoying. Even with the mcl amps one ends up needing to crank the attenuators right up to cancel out the bulk of the pre-amp gain.

Maybe I’m wrong but I’d guess that the pre-amp is only useful in a minority of cases. So it doesn’t feel like it’s worth having on the main population variant as it’s a bit niche.

hartytp, 2019-12-04:
Do you really want to burn off an extra >1W and inflate the bom for something that’s sometimes mildly convenient?

jordens, 2019-12-04:
If we want decent 0dBm output power I don't see any alternative to a preamp. I'm open to a design that achieves that while giving some other advantage, e.g. using less power. But I suspect that it'll be difficult. Not needing an external attenuator is not a sufficient reason IMO.

jordens, 2019-12-04:
Maybe the smartest solution is to look at the transformer and get 3 dB or 6 dB for free out of there. Then we could skip the preamp.

jordens, 2019-12-04:
Oh. By the way. The AD9912 variant has an error. Without the split 50R before the balun, we need to use a 2:1 balun.
And we should and can do that for the AD9910 as well. Just straight into a 2:1 balun and get 3 dB.

hartytp, 2019-12-04:
> Maybe the smartest solution is to look at the transformer and get 3 dB or 6 dB for free out of there. Then we could skip the preamp.

> And we should and can do that for the AD9910 as well. Just straight into a 2:1 balun and get 3 dB. Same compliance, same current.

Good idea. That would work nicely for me. Then 

> Not needing an external attenuator is not a sufficient reason IMO.

Generally, the arguments against the pre-amp are:
- noise: AFAICT we loose 9dB of NF due to the 9dB attenuator needed to reduce the pre-amp gain (i.e. from the data sheet numbers I do not believe the DDS is the limitation if operated without the PLL)
- Power consumption
- Stability: pre-amps generate heat and cause drifts/distortion
- Cost/complexity

What output power do people actually want? Is the target 0dBm?

hartytp, 2019-12-04:
> Someone up for some testing? Just bypass the preamp, put a 2:1 balun in there, remove the resistors before the balun and measure output power and ip3 at 3, 30, 300 MHz.

Yep. I'm very happy to test this. So, the proposal is to:
- bypass the pre-amp (I'll remove it and solder an 0850 0R resistor across the pads)
- remove R38 and R54
- Replace TR1B (ADT1-1WT) with a 2:1 balun. Any recommendations for part number?

Then measure what? Output power and harmonics with, say, a 100MHz carrier?

hartytp, 2019-12-04:
I won't be able to do this for a week or so (waiting for a new batch of boards), but we're not in a hurry to make this change anyway...

jordens, 2019-12-04:
Just bypass the preamp, put a 2:1 balun in there, remove the resistors before the balun and measure output power and ip3 at 3, 30, 300 MHz. Either TC2-1T (cheaper, smaller, would be usefule for the clock inputs as well) or ADT2-1T (same footprint).

jordens, 2019-12-04:
And if you are adventurous figure why the DAC34H84 can also operate with a 4:1 balun, resistors before the balun. And since they are all three nominally the same DAC specs, we could give the TC4-1T/ADT4-1T a go as well.

hartytp, 2019-12-04:
> Just bypass the preamp, put a 2:1 balun in there, remove the resistors before the balun and measure output power and ip3 at 3, 30, 300 MHz. Either TC2-1T (cheaper, smaller, would be usefule for the clock inputs as well) or ADT2-1T (same footprint).

Okay. I'll do that before Christmas.

hartytp, 2019-12-04:
> And if you are adventurous figure why the DAC34H84 can also operate with a 4:1 balun, resistors before the balun. And since they are all three nominally the same DAC specs, we could give the TC4-1T/ADT4-1T a go as well.

Not sure I'm the right person to answer that one. @gkasprow ?

dhslichter, 2019-12-05:
FWIW I would prefer that we leave the pads in place for the preamp as it currently is and have a solder-jumper option for bypass.  People who don't want a preamp can have it stuffed without those components, people who do want a preamp (this may be us in a variety of instances) can keep it.  I don't see this as something people want/need to switch on the fly.  Thoughts?

hartytp, 2019-12-31:
Closing this as, on reflection, Urukul is now mature enough that we should prioritize minimizing changes over marginal design tweaks; the current situation is fine.

closed 2019-12-31 by hartytp