sinara-hw/Urukul #46: Fix washer stack up

hartytp, 2020-02-18:
https://github.com/sinara-hw/meta/issues/4
https://github.com/sinara-hw/Urukul/issues/45#issuecomment-587478088

gkasprow, 2020-02-20:
what if we design and 3D print dedicated spacers that will fill the gap between the panel and the SMA? In this way, it will provide mechanical fixing to the panel without the need of using the nut.

jordens, 2020-02-21:
Yes. Just a sleeve should be fine IMO. Like one of the original white plastic washers on the inside between SMA and panel, just with a bit more depth to be flush with the outside of the panel.

gkasprow, 2020-04-30:
Here we are.
[image]
[image]

gkasprow, 2020-04-30:
the files are here:

[MECH.zip](https://github.com/sinara-hw/Urukul/files/4561704/MECH.zip)

gkasprow, 2021-02-19:
we use 3D washers for some time so let's close it.

closed 2021-02-19 by gkasprow