sinara-hw/Urukul #47: Consider adding TLT balun foot print

hartytp, 2020-02-18:
Optional to consider for v1.5:

> We've modified a couple boards to use the third Nyquist image. That works fine apart from the balun being out of its depth. It would be nice to have a alternative footprint for a TLT balun in that standard smaller case style.

jordens, 2020-02-18:
[TC1-1T-152X+](https://www.minicircuits.com/pdfs/TC1-1T-152X+.pdf) I think

dhslichter, 2020-02-18:
+1 to this idea in general.  @jordens is the reason for choosing this part just for pin compatibility with existing pads (i.e. no change in copper necessary)?

jordens, 2020-02-18:
It's actually different pads than the current footprint. But that case and pinout is pretty generic from what I could tell and would allow going to a cheaper/more commonly available package for the standard 1-500 MHz balun style as well.
That TLT part number is the only one that I could find that would work with the required biasing topology straight away (center tap).

dhslichter, 2020-02-18:
p.s. the existing balun ADT1-1WT+ is a 75 ohm nominal balun...given that we're playing fast and loose with nominal impedances, has anyone tried dropping in an ADTL2-18+, which is pin compatible, nominal 50 ohm impedance but 2x transform, to see if the output is reasonable?

dhslichter, 2020-02-18:
@jordens we need a center tap bias?  I guess I missed that.

dhslichter, 2020-02-18:
Or is that just for the 9912, seems that the 9910 sets its dc level using the load resistors, right?

jordens, 2020-02-18:
Both are center tap grounded on the eval boards. See #40, #39, #43 for more info and questions on that topic.

jordens, 2020-02-18:
Maybe there are other topologies that don't blow up at DC (i.e. when FTW=0) and still bias/terminate fine.

dhslichter, 2020-02-18:
> Both are center tap grounded on the eval boards. 

Weird.  Anyway, I am not trying to come in and blow things up, nor is there time for discussion on this given the urgency of putting at v1.5 release out today.  I'm just a little curious/surprised at AD for the eval board choices for the reasons discussed in #40

gkasprow, 2020-02-19:
dual footprint is not that trivial here. Let's drop it for v1.5
[image]

jordens, 2020-02-19:
Ack

hartytp, 2020-02-19:
Does anyone want to keep this open or shall we close?

jordens, 2020-02-19:
We should keep this open.

gkasprow, 2021-02-19:
I managed to slightly recuce the original balun pad width and place the second TLT balun
[image]

closed 2021-02-19 by gkasprow