sinara-hw/Urukul #49: Output un-uns

hartytp, 2020-02-18:
Only if this can be done without major rework or compromises on the layout: add a footprint for DNPd un-uns (e.g. TC1-1(T)(X)(+)) on the outputs. The idea of these (please add a schematic annotation documenting this) is to provide the possibility of breaking ground loops (in that case the "grounding" capacitors should be DNPd or possibly even just removed from the design).

gkasprow, 2020-02-19:
[image]
[image]

closed 2020-02-19 by gkasprow

hartytp, 2020-02-19:
Lgtm. Only comment is that I wonder if a bigger resistor package for the 0R would keep the gap between the input and output pins of the un un a bit wider. I don’t see this causing problems in practice however