sinara-hw/Urukul #51: Support cPCIs backplane

jordens, 2020-02-19:
Place jumpers between MMCX clock in and Samtec hermaphroditic (https://github.com/sinara-hw/Phaser/issues/74#issue-527610429)

gkasprow, 2020-02-19:
Adding the Samtec connector is a lot of work. The CPLD mus be moved and re-routed. Let's leave it for the next release.
[image]

jordens, 2020-02-19:
Sure. Just wanted to put it on the list for the next revision.

gkasprow, 2020-02-19:
btw, this is how the CPCIS version of Urukul will look like
https://autode.sk/38NAshC

gkasprow, 2021-02-19:
We don't have many boards with these clock connectors, I decided to move it close to EEM connectors to save some space. Next Phaser/Mirny release will have it also moved closer to the edge.

closed 2021-02-19 by gkasprow