sinara-hw/Urukul #52: C155

hartytp, 2020-03-21:
@gkasprow given our experience with Fastino, do you think C155 (LT3045 output decoupling 22uF, X5R 16V, 0805) will be sufficient? That LDO needs 10uF min to ensure stability, so I wonder if this could be a bit marginal for the 7V output? My experience with other boards has been that when the capacitor is marginal some boards exhibit instability while others are fine, so it's best to err on the side of caution.

gkasprow, 2020-03-21:
LEt's simply use higher capacitance

hartytp, 2020-03-21:
Sounds good so long as we end up with >10uF after accounting for tolerance and bias.

As this is a bom change only and no new boards have been stuffed yet can we ask creotech and technoststem to make the change before shipping and also update the release?

gkasprow, 2020-04-02:
we can change to 33uF without modifyig the PCB

dtcallcock, 2020-06-19:
Can this be closed @gkasprow as it was changed in v1.5?

gkasprow, 2020-06-19:
Yes, close it

gkasprow, 2020-06-26:
In the new release, let's use 10+22uF or 33uF in bigger package. 33uF is difficult to obtain and increases the BOM count.

gkasprow, 2021-02-19:
I added another 22uF capacitor

closed 2021-02-19 by gkasprow