sinara-hw/Urukul #53: Question about current PCB stackup

HarryMakes, 2020-11-11:
I would like to obtain the following information for Urukul, similar to my post for Kasli: https://github.com/sinara-hw/Kasli/issues/86

1. Are the current Gerber files showing the correct PCB stackup? E.g. has all the impedance control been reflected by the output files?

2. NCAB provides certain guidelines for designing PCBs ([link](https://www.ncabgroup.com/pcb-design-guidelines/)). Is the current Urukul PCB design complying with their standards, if any?

gkasprow, 2020-11-11:
1. buy Altium and Hyperlynx licenses and stop asking about effects of modifications
2. If you believe you could design it better, with NCAB guidelines, do it and stop asking stupid questions. This is open hardware design, feel free to modify it and later on take responsibility for the product.
3. there are already 2 vendors of the modules and they managed to solve such issues themself. I've already put a lot of effort into RnD and don't have time for changing the design and discuss it because you don't like it.
4. if you find some merit issues to discuss, i'd happy to do it. But comments about compliance of the design with some guidlines don't add anything to the discussion. Simply - if you find some bug, open an issue and I will try to solve it. If you want to change design, do it on your own risk. Im supporting nearly 40 designs and have better things to do than discussing production issues.

closed 2020-11-11 by gkasprow

gkasprow, 2020-11-11:
Creotech and Technosystem are the companies that know much more about the production of these modules. Contact them and discuss the issues. They can modify the design and run the simulations.

HarryMakes, 2020-11-13:
Greg, thank you for the comments. My questions on both here and my other post on Kasli were made to @sbourdeauducq's request in order to understand the stackup for both boards, and I have no intention to challenge any decisions you made during the design process. I was tasked to open these issues with the only goal of enquiring about impacts of modifications or adjustments proposed by the manufacturer rather than myself. I have little experience in PCB manufacturing so my questions might sound simple or even naive. But for future reference about the production cycle in general, I would like to document a summary of what I learnt from [the discussion](https://github.com/sinara-hw/Kasli/issues/86).

I learnt that it has been made clear the Gerber files residing in the repos of both Kasli and Urukul are correct and followed for production at CTI and TS such that impedance control is done properly. This should also apply to all aspects of the PCB design like choices of components parts, meaning that switching to other manufacturers means there is need to re-evaluate all effects due to differences in vendors, materials, techniques, etc.

I mistook the NCAB guidelines for PCB manufacturing "standards"; rather, they are simply a guide for PCB designers in case their Gerbers are still prone to mistakes, which of course is not useful given maturity of the projects. Besides, from our discussion it is clear that any high-quality PCB manufacturer would provide a "default" stackup with impedance profiles, which whoever is making an order should check with the schematics. In the end, it is an engineer's responsibility to verify whether or not modifications made by the manufacturer would satisfy the design requirements.

Finally, as a software developer myself I am not capable of proposing fool-proof solutions to hardware bugs or taking full responsibility into such. However, I will keep trying my best to find and document possible hardware issues in a concise manner for all Sinara boards. Again, thank you for letting me understand all kinds of considerations to make in actual production.

gkasprow, 2020-11-13:
I explained a lot more production issues in Kasli thread.

HarryMakes, 2020-11-13:
@gkasprow Yes, thank you very much for all the comments and everyone's involvement in the thread. Greg, sorry once again about my poor phrasing and misunderstanding in my questions!