sinara-hw/Urukul #56: v1.5.1 changelog

gkasprow, 2021-02-19:
It's mostly cosmetic, with no changes to functionality so let's leave the same HW REV and call c1.5.1
- changed washer PN #46 
- added alt TLT balun option #47
- added clock connector (DNP) #51
- added another 22uF cap #52
- J2 aligned #55

SingularitySurfer, 2021-10-04:
Hi @gkasprow, as far as I can tell the alternative footprint option from #47 is not actually in v1.5.1 and 1.5.2. I still see the old footprint. Or am I missing something?

gkasprow, 2021-10-04:
Dual footprint is present in 1.5.x
[image]

SingularitySurfer, 2021-10-05:
Ah, I think the problem is that the newest releases are tagged 1.5rc1 and 1.5rc2 and not 1.5.1 and 1.5.2 (which is also Production right now). In master the footprints are updated.

jordens, 2021-10-05:
@gkasprow could you tag 1.5.2 in the repository and upload corresponding PDFs and board output zips to that release?

jordens, 2021-10-06:
Thanks. No problem.

closed 2023-06-23 by marmeladapk