sinara-hw/Urukul #57: DC floating section

dtcallcock, 2021-03-29:
The section between C49 and C50 has an undefined DC potential.  I always thought this was to be avoided. Though I can't find a good reference for why this is bad right now  (dC/dV of caps?, making an antenna?), so I am happy for someone to tell me it doesn't matter.  Solution would be a BOM change to make default C50 population a 0R.

[image]

gkasprow, 2021-03-29:
At least such configuration cannot be simulated using SPICE:)
I don't know any mechanism that caps could build-up DC voltage causing significant change to their capacitance.

dhslichter, 2021-03-31:
agreed that default BOM change to make C50 a 0R would be the way to go here.

dtcallcock, 2021-03-31:
Despite my failure to articulate much beyond 'it gives me bad feelings', I don't think there is any good reason **TO** have two caps in series like this.  But obviously can wait until BOM needs changing anyway.

gkasprow, 2021-08-13:
fixed

closed 2021-08-13 by gkasprow