sinara-hw/Urukul #58: Attenuator DC block cap

dtcallcock, 2021-03-29:
There is no DC blocking cap on input of IC3.

[image]

From IC3 [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/hmc542b.pdf):

[image]

Was this an oversight or a does the DC GND path provided by FL2+TR1 mean we don't need this (though if this is the case, you could probably make a similar argument for the output blocking cap given the next component is an attenuator that has a DC path to GND)?

gkasprow, 2021-03-29:
Good point. Some switches have an internal charge pump and can work with GND-referenced signals. In our case lack of caps can induce nonliearities.

jordens, 2021-03-29:
The jumper decouples.

dtcallcock, 2021-03-29:
Ok, now I spot the reference to 'C28' in this comment.

[image]

But AFAICT, C28's location or value aren't on the schematic.  Can the ambiguities of the 'mystery red box' easily be fixed?  If not then more clarity in the comment would help.

gkasprow, 2021-08-13:
I added a comment 
[image]

gkasprow, 2021-08-13:
I also changed the frame around components so the values will be visible in pdf

closed 2021-08-13 by gkasprow