sinara-hw/Urukul #59: RF output ESD protection

dtcallcock, 2021-03-31:
We killed a switch on an Urukul.  This is the final active component in the RF chain so I'm guessing it was from ESD.  

Have there been other reports?  Was adding ESD protection to the RF outputs ever discussed? 

We could add an RF TSV at R4 with just a BOM change.  You can get [TSVs with like 0.04pf @250MHz](https://media.digikey.com/pdf/Data%20Sheets/Littelfuse%20PDFs/AXGD1_Series.pdf).

dhslichter, 2021-03-31:
Could you solder one on to an Urukul output and see if it changes the harmonic distortion of the output signal?  Basically a test before/after at different powers and frequencies.  These components you mentioned look like they should basically have no effect on signal integrity but would help with ESD issues.

dtcallcock, 2021-03-31:
Yes, I had it in my head that this would be much more complicated.  Am I missing something about the limitations of this kind of TSV or what's needed here @gkasprow ?

Happy to test if those 10G eye diagrams in the data sheet aren't convincing!

dhslichter, 2021-04-01:
I mean, it looks like it's exactly what one wants, but it would make sense to buy a few and take 20 minutes to solder one on and sanity check that nothing goes haywire before putting it on the BOM.  Also be useful to know whether they do the job of protecting the switch.  This could be validated with a few switch eval boards, a grad student with slippers and shag carpeting, and a couple of these TSVs 😁

gkasprow, 2021-08-13:
I added [TVS](https://wwanw.eaton.com/content/dam/eaton/products/electronic-components/resources/data-sheet/eaton-0402esda-mlp-esd-suppressor-data-sheet.pdf) 
If we notice issues with production tests, they will be removed

closed 2021-08-13 by gkasprow