sinara-hw/Urukul #60: IC22 voltage

gkasprow, 2021-04-27:
I've just noticed that IC22 has a "0.8V" annotation close to the FB input.
In fact, it should be 0.7V. The TPS62148 is able to deliver output from 0.8V but the nominal FB voltage is 0.7V
With current resistors, it produces 6.5 instead of 7.5V which makes the 7V LDO out of regulation.
Can somebody who has Urukul measure the voltage on C211?
[image]

jordens, 2021-04-27:
Isn't this https://github.com/sinara-hw/Urukul/issues/20 ?

atcher0, 2021-08-09:
Voltage measured on C211: 6.58 V (I checked on two different Urukul AD9910 v1.5.1 boards). It produces about 6.33 V from the IC21 LDO.

gkasprow, 2021-08-13:
R134=75k
R136=7k5

gkasprow, 2021-08-13:
fixed in v1.5.2

closed 2021-08-13 by gkasprow

dnadlinger, 2021-08-13:
What's the practical consequence of this being left unpatched? Worse output power stability?