sinara-hw/Urukul #64: significant harmonic content observed from emitted signal (AD9910)

collineps, 2021-11-30:
We are observing significant harmonic distortion in signals emitted from our Urukul/AD9910 board.
The distortion can be up to -20 dB relative to the fundamental frequency, which is unacceptable for my lab's applications.
The distortion occurs on every port, for a wide range of frequencies, and is not adequately suppressed by:
* Attenuation through external hardware
* Attenuation through ARTIQ software commands (with the exception of extremely high attenuation)
* Modification of the AD9910 chip auxiliary DAC register value
* Modification of the AD9910 chip amplitude scale factor register value

Our system is clocked by a Kasli card as part of an integrated system delivered by a vendor, and our measurements are performed with multiple signal analyzers that terminate with a 50 ohm load.

Please see the plot below for an example of the harmonic content we are observing:
[image]

I have several questions concerning this distortion:
* Does the Urukul have a formal specification for the expected harmonic content? For example, the AD9910 specifies very low harmonic distortion around 200 MHz
[image]:
* Is this issue with the Urukul boards known? Is it replicable?
* Is it possible to obtain a cleaner signal from the Urukul? If so, how?

We believe the harmonic content indicates that a component in the board is being saturated.
Our experiments show non-linear (but inadequate) suppression of the harmonic peaks using 'software' attenuation and auxiliary DAC register values, suggesting that the saturated component is downstream of both components.
From our initial investigation, it looks like the ERA-3SM+ amplifier at the end of the circuit may be being saturated.
Our calculation of the expected signal power at that amplifier is about an order of magnitude greater than the expected value indicated by a note in the Urukul drawings:

> With about 1dBm out of the DDS, 0.5 dB insertion loss from the Balun, 0.5 dB from the lowpass, 1.5 dB from the attenuator, we need a 9dB T-pad to attenuate that before the ERA-3+ with 23 dB gain and P1dB of 13 dBm at our frequencies.

The difference is due to an order of magnitude greater power being emitted from the DDS chip, which agrees with our measurement of the signal power.
Additionally, the greater signal power we believe that the amplifier is experiencing does agree with the saturated power range listed in the ERA-3SM+ amplifier specifications.

I appreciate any insight you have into this problem!
Thank you!

gkasprow, 2021-11-30:
Urukul was designed for aplications that do not require low harmonic content. It seems you are the first person who needs better performance.
However, it is possible to lower the THD by removing the amplifier. The PCB has such assembly option - remove the amplifier, inductor and install 0603 0R resistor instead.
Than use external low THD amplifier such as i.e. Booster_HL. [It's a Booster version with high linearity power stage](https://github.com/sinara-hw/Booster/issues/385).
[image]

gkasprow, 2021-11-30:
If there is sufficient interest, we can make a low THD version of Urukul.

jordens, 2021-11-30:
The harmonics [were](https://github.com/sinara-hw/sinara/issues/354#issuecomment-352859041) and are well known. The system was designed like this and is behaving accordingly. To get low harmonics, go to low powers, remove the amplifier and potentially the other components (balun, switch, attenuator, pad) in the chain, lower the DAC bias, and also just filter the output.
If you need an ultra-linear medium-power agile RF source, then the design would look different.
I'm also unsure your understanding of P1dB and the overall ERA datasheet is correct.

collineps, 2021-12-03:
I understand, thank you for your quick and detailed responses. We will try the suggestion of bypassing the board amplifier and take another look at our system design.