sinara-hw/Urukul #68: 1GHz crosstalk

sbourdeauducq, 2022-05-25:
Reported by Andrew Van Horn at Duke:
"A few months ago, we observed a 1GHz crosstalk on all four output channels of the Urukul 9910 v1.5 and 9912 v1.3 when a 1GHz external clock was connected. This crosstalk occurs independent of the clock source and clock reference connection (SMA or MMCX).

I have attached four scope captures which show the presence of this crosstalk on the Urukul 9910 v1.5 and 9912 v1.3 when using the internal clock reference. The crosstalk is also present when using the 1GHz as the reference. 

We observed the 1GHz reference resulted in lower overall phase noise as expected, so in our systems we plan to use the 1GHz clock reference with a 400MHz LPF on the output of the Urukul to mitigate the effects of this crosstalk. " 
[image]
[image]
[image]
[image]

sbourdeauducq, 2022-05-25:
Urukul_v1 5_9910 400MHz_Output With_Filter
[image]

Urukul_v1 5_9910 400MHz_Output No_Filter
[image]

Urukul_v1 5_9910 200MHz_Output With_Filter
[image]

Urukul_v1 5_9910 200MHz_Output No_Filter
[image]

Urukul_v1 5_9910 100MHz_Output With_Filter
[image]

Urukul_v1 5_9910 100MHz_Output No_Filter
[image]

jordens, 2022-05-25:
Check whether the clock in sma connector and the RF out sma connectors are well clear of the panel everywhere. Also their shields. That's where it typically couples.

sbourdeauducq, 2022-06-01:
"I am quite confident that is not the major source of the issue we are seeing. 

First, the cables were kept as far apart from each other and from the front panel as the geometry of the front panel would allow. 
Second, by configuring the Urukul to use the internal clock source and monitoring the output with a spectrum analyzer, the 1GHz noise only appears when the 1GHz clock is connected to the clock SMA. While I am screwing in the 1GHz SMA, the noise abruptly appears when the signal pin makes contact. "

jordens, 2022-06-01:
The panel and the connectors in that panel. Nothing to do with the cables. It's a manufacturing problem. Has been pointed out numerous times.
Feels awkward to help in this way.

andrewvh4, 2022-06-01:
@jordens Is this manufacturing problem fixed in the next revision of the Urukul? And is there any way to modify current revisions to prevent this noise?

jordens, 2022-06-02:
Afaik no. But I don't know what the next revision is.

See the advice in my initial comment above.

If you don't find it here or on phaser in the meantime I'll try searching it for you next week.

andrewvh4, 2022-06-07:
I modified the Urukul by removing the front panel altogether to eliminate any coupling through the front panel. This eliminated the 1GHz crosstalk.

Internal 125MHz clock, 200MHz output, no front panel
[image]

External 1GHz clock, 200MHz output, no front panel
[image]

gkasprow, 2022-09-17:
Do you mean electrical connection of the SMAs and the panel? We use 3D printed distances to make sure isolation is there.

gkasprow, 2023-05-23:
We confirmed the results. The issue is caused by impedance of optional AC-coupling (isolation) RCs. After we shorted SMA connector to nearby GND with thick wire, the crosstal got reduced by a few dB
In the next revision I will short SMA to GND using 4 thick traces. AC-coupling would require cutting the trace.

gkasprow, 2023-06-21:
fixed in v1.5.5

marmeladapk, 2023-06-23:
@gkasprow Isolation of SMAs also requires removal of resistors on top layer. I think that description on bottom overlay should reflect this.

gkasprow, 2023-06-23:
done

closed 2023-09-07 by maciejprzybysz