sinara-hw/Urukul #69: artiq phase noise consultation

zhouyabin, 2022-06-23:
I used a low phase noise signal source as my artiq  frequency reference，The reference frequency is 100MHz
[image]
[image]
this source looks relatively clean.
But my artiq DDS output frequency produces some other unknown noise besides the sidebands. And his phase noise curve is not very good
[image]
[image]
He looks bad, the frequency of the output produces other noise besides the sidebands.
Below is the frequency output and phase noise curve using artiq internal clock reference
[image]
[image]
After artiq connects the external clean source as the frequency reference, the phase noise curve should be better, but it doesn't seem to be, I don't know how to solve this problem, thank you all.

gkasprow, 2022-07-31:
Can you describe the setup you are testing?

gkasprow, 2023-02-09:
I tested early versions of Urukul and no such effects were observed.  Due to component shortages we replaced some parts and it's quite possible that it caused the issue.

aewallin, 2024-04-08:
FWIW here is some phase-noise data (from 2022) from our Urukul (AD9912) at ca 25 MHz output. This is measured with a Microsemi 3120A phase-noise-probe, so the max measurement frequency is 30MHz. The reference-frequency for ARTIQ and the phase-noise-probe comes from the same source (Hydrogen maser) via a distribution amplifier.

[image]