sinara-hw/Urukul #72: Urukul layout

maciejprzybysz, 2023-03-27:
We made some investigation on Urukul v1.5.3 and found some issues on layout and signal integrity.

1. DDS0_SYNC_IN/OUT signals (layer L3) have poor reference on L2 layer. 

   [image]

2. Also other DDSx_SYNC_x signals have problem with reference, due to specifics of stackup. In general L2 should be reference for L1(TOP) and L3 signal layers, P5 should be reference for L4 and L6(BOT). In fact dielectric thickness between L3 and L4 is much lower than between L2/L3 or L4/L5, so there is strong coupling between 2 center layers. 

   [image]

   DDSx_SYNC_x signals on layer L4 on the background of layer L3 looks like this:
   [image]
   [image]
   [image]
   [image]

   This could cause step changes in impedance and signal integrity problems.
<br>

3. ORIGIN. This issues affect Urukul v1.4 and later.

   Till v1.3 L2 was mostly GND split plane, there wasn't any signals there, so DDS0_SYNC_IN/OUT had good reference.
   [image]

   In addition, there was no GND plane on L3 and L4 layers, so the only reference for DDSx_SYNC_x signals was on L2 or L5 layer. 
[image]

gkasprow, 2023-03-27:
that's strange. Usually when designing 6 layer boards I treat them as two 3-layer boards having L3-L4 separation dominating so L3 and L4 impedances are defined by thin dielectric to L2 and L5.
These issues can be easily fixed by changing stackup and re-routing these two lines which don't have valid reference on L2.
I don't think however that this is the main issue why we don't have synchronization.  in v1.4 we swtiched to another CPLD in bigger package

gkasprow, 2023-06-21:
fixed in v1.5.5

marmeladapk, 2023-06-23:
Mostly fixed, remaining problems are in a separate issues.

closed 2023-06-23 by marmeladapk