sinara-hw/Urukul #73: Problems with channel 0 synchronization

marmeladapk, 2023-06-20:
6 attenuator SPI lines run under PLL filter of AD9910 in channel 0:

- ATT_S_OUT0
- ATT_S_IN1
- ATT_S_OUT1
- ATT_S_OUT2
- ATT_LE2
- ATT_S_IN2

These lines seem to have a significant enough effect on PLL filter that it causes SMP ERR to appear and sometimes even phase glitches in output signal (caused by ATT_S_IN2).

To fix this, 
- these lines must be rerouted away from loop filter and P1V8_CLK_VDDA (maybe even away from DDS0 if it's possible)
- stackup issues must be fixed (https://github.com/sinara-hw/Urukul/issues/72)
- as well as these signals should have a proper continuous reference plane.

More info can be found in https://github.com/m-labs/artiq/issues/1692.

gkasprow, 2023-06-21:
fixed in v1.5.5

marmeladapk, 2023-06-23:
I'd say that these lines should be rerouted away from P1V8_CLK_VDDA under AD9912 as well.
[image]

Is it possible or worth it to fight for continuous reference plane for these signals?

gkasprow, 2023-06-23:
I think we can do it

gkasprow, 2023-06-23:
I did what I could

dnadlinger, 2023-06-25:
Given that the PLL nodes are apparently more susceptible than anticipated, I suppose it might be interesting to see whether togglign all the potential agressor lines as aggressively as possible leads to phase noise on the output.

closed 2023-06-26 by marmeladapk