sinara-hw/Urukul #80: v1.5.6 changelog

maciejprzybysz, 2023-07-13:
- Impedance profile added
- Additional directives for 50ohm lines added
- Trace width unified for impedance control
- P_version project parameter added, rev on schematic table filled with version number
- PCB/Urukul.PCBDwf file with stackup and impedance table added
- Impedance table added on PCB Mechanical 1 layer
- Minor cosmetic changes on SCH

dnadlinger, 2023-07-14:
Wait, https://github.com/sinara-hw/Urukul/issues/74 is not fixed in 1.5.6?

maciejprzybysz, 2023-07-14:
It was fixed in v1.5.5, I just added Impedance control in new version.

maciejprzybysz, 2023-07-14:
v1.5.5 changelog is missing, @marmeladapk could you make issue with v1.5.5 changes to have history tracking? IDK if this is the best place, but it's much better than commit message.