sinara-hw/Urukul #82: CPLD XC2C256-6FTG256C is EOL

MorganTL, 2024-05-23:


gkasprow, 2024-05-23:
any suggestion what family we could use here?

Spaqin, 2024-05-24:
Mirny uses the same CPLD.

A suggestion would be to use Artix-7, like XC7A35TICSG324 or XC7A100TCSG324, depending if 35k logic cells would be fine or we need 100k.

gkasprow, 2024-05-24:
Artix is FPGA, which means it has much higher internal delays and is suitable for synchronous designs. CPLD was placed here as a simple asynchronous glue logic. Such a big FPGA would significantly increase the board cost because we also need extra power rails.

gkasprow, 2024-05-24:
Since AMD and Lattice do not make any CPLDs anymore, we can move to Cypres or Altera:
[5M2210ZF256I5N](https://www.digikey.pl/pl/products/detail/intel/5M2210ZF256I5N/2660801)
[EPM570F256C4N](https://www.digikey.pl/pl/products/detail/intel/EPM570F256C4N/705413)
[CY37256VP256-100BGC](https://www.digikey.pl/pl/products/detail/rochester-electronics-llc/CY37256VP256-100BGC/11512522)

@jordens Maybe the ICE40 family could be fine here?

jordens, 2024-05-24:
That would be interesting to explore.

kaolpr, 2024-06-06:
If there are no pending ideas on implementing some more elaborate logic we could go with iCE40 HX8K as in Fastino. This would let us drop LVDS transceivers too (same for Mirny).

dnadlinger, 2024-06-06:
The biggest (in relative terms) design we currently need is the status readback for SUServo mode: https://github.com/quartiq/urukul/pull/11. For this, we ran into size limits of the parts that were on older revisions, though there is some discussion on how to best implement this in a glitch-free way still unresolved (see linked PR).

jordens, 2024-08-14:
Is anybody working on this? I have no idea about stock situation at cti/ts. Given the revision/production/development/testing lead time this may be something that should have been started a while ago.

gkasprow, 2024-08-14:
Yes. I expect to have Mirny and Urukul designs done in 2 months. Two guys are working on these designs.

gkasprow, 2024-08-14:
CTI secured stocks for next year

jordens, 2024-08-14:
Who is doing the gateware redesign? That should be done first to verify/decide the choice of chip (did you choose already?).

gkasprow, 2024-08-14:
we will go for same chip as in Fastino

pkozakiewicz, 2024-08-19:
> I have no idea about stock situation at cti/ts.

We are well stocked with CPLDs and finished devices so this situation does not threaten availability.