sinara-hw/Urukul #83: DIOT version

gkasprow, 2024-08-31:
A new DIOT/EEM version is on the way:
- take the existing design and replace CPLD with ICE40 - circuit and EEM connectivity copied from Fastino-DIOT (with proper power sequencing)
- get rid of LVDS translators
- move power supplies away from DDS chips, we have a lot of board area
- make sure all points are checked: https://github.com/sinara-hw/DIOT-Template/issues/1

newell, 2024-12-29:
I noticed on the new Urukul board that there is a 25 MHz clock (CLK_25MHz in the schematic) tied to pin K9, which for the part is of type GBIN.  Is this going to be the FGPA's dedicated clock?

gkasprow, 2024-12-29:
There are more issues - the DIOT clock is not routed to the clock distribution IC.
The 25MHz clock is just copied from Fastino design; we would prefer to get 100MHz DRTIO clock from the backplane