sinara-hw/Urukul #85: The board picks up smartphone communication signals

jzli559, 2024-12-20:
I observed that the Urukul board can pick up signals from smartphone communications. Using an oscilloscope, I probed the signal from the SMA connector on the front panel of the board. Throughout the process, I didn't set any output of the board. When I made a voice or video call on a smartphone and brought the phone close to the SMA connector, I observed high-frequency (over 1 GHz) signals on the oscilloscope. After turning off the call, the signal returned near zero.

I did the test under different configurations, and the results were similar.
- Urukul installed in a crate: connected to a Kasli that is powered and grounded.
- Urukul installed in a crate: connected to a Kasli that is not powered but the crate is grounded.
- Urukul installed in a crate: connected to a Kasli, not powered and not grounded.
- Urukul isolated: not connected, not powered, and not grounded.

I did further tests with a spectrum analyzer (FSV3030), and the results are as follows:

| [image] | [image] |
| In crate, grounded, not powered; no phone | In crate, grounded, not powered; with phone (5 cm distance) |
| [image] | [image] |
| In crate, not grounded, not powered; no phone | In crate, not grounded, not powered; with phone (5 cm distance) |

After the tests above, I additionally tested the noise by setting the channel to output a signal (100 MHz, 0 dB att, 1.0 amp).

| [image] | [image] |
| 100 MHz, 0 dB att, 1.0 amp, no phone | 100 MHz, 0 dB att, 1.0 amp, with phone (5 cm distance) |

For simplicity, all six images above were tested on channel 0. The only related configurations are the ground and the phone. The results of others are similar. Some of the noise peaks were related to the ground, but M4 and M8 were clearly related to the phone. until the distance was increased to about 3 m.

This looks like the board doesn't specifically handle noise over 1 GHz, and the wavelengths of some of the phone signals are similar to the length of some wires on the board.

Similar phenomena were observed when testing other cards in the same crate, including the Kasli J3, Clocker Output 9, and all channels of the DIO-BNC card. Other interfaces haven't been tested yet.

gkasprow, 2024-12-20:
pls do the following test:
- use the same RF cable. 
- place 50Ohm termination and then short SMA nut with unpowered, grounded crate. Beware of alu oxidation layer - make sure you attach the SMA terminator to some steel screw/
- measure the spectrum with the phone as above
- do not place 50Ohm termination and then short SMA nut with unpowered, grounded crate
- measure spectrum with the phone as above

jzli559, 2024-12-21:
Before anything else, I need to correct the test results I posted earlier. Both the first and second images were powered and grounded, with no DDS outputs set.

Based on your suggestion, I did some new tests. First, the spectrum analyzer (SA) requires grounded during using, and it shares the same ground with the crate. Therefore, the SMA nut of the RF cable is inherently grounded with the crate without being shorted again. I'm not sure if connecting the RF cable from the SA to the Urukul's SMA would create a ground loop.

Below are the new test results.

(Remeasured) In crate, grounded, not powered; no phone:
[image]

(Remeasured) In crate, grounded, not powered; with phone (5 cm distance):
[image]
[image]
[image]

All the following tests are also within crate, grounded, not powered.

Urukul ch0 connected to a 50 Ohm terminator; SA probe connected to noting or a 50 Ohm; phone close to ch0:
almost no signal.

Urukul ch0 connected to nothing; SA probe connected to noting or a 50 Ohm; phone close to ch0:
almost no signal.

Urukul ch0 connected to nothing or a 50 Ohm; SA probe connected to noting; phone close to SA probe:
almost no signal.

Urukul ch0 connected to nothing; SA probe connected to a 50 Ohm; phone close to SA probe:
(Urukul ch0 connected to a 50 Ohm was similar)
[image]

The last image also showed some clear noise signals but not as big as connected to ch0.

gkasprow, 2024-12-21:
Is the front panel grounded?

jzli559, 2024-12-21:
> Is the front panel grounded?

Yes. The card is installed in the crate. I measured with a multimeter and found that the outer shell of the SMA connector on the front panel is electrically connected to the crate.

gkasprow, 2024-12-21:
That's not enough. You checked DC path but we are interested in AC path. Just check if the grounding resistors are soldered.

jzli559, 2024-12-22:
Do you mean using the multimeter to test resistors R179 ~ R182?

gkasprow, 2024-12-22:
There are 2 resistors amd 2 capacitors that comnect panel with PCB ground