sinara-hw/Urukul #86: Urukul DIOT: FPGA CRESET is held down on power up, preventing the FPGA from booting

Spaqin, 2025-04-25:
I was having a difficult time flashing/running the bitstream - tried with FTDI for SPI and with kasli-i2c, and it took me a while to figure out why.

Verified with a logic analyzer that FPGA CRESET is held down after initial power up:

[image]

I noticed that flashing with ``iceprog`` and FT232H may seem to fail (no readback), but it did work, as with kasli-i2c I could toggle the FLASH_UP_EN pin and then the bitstream that I had flashed earlier would be finally loaded.

Traced it on the board. The board I have next to me uses SC18IS606PWJ rather than SC18IS602B (like Fastino would).

[image]

It does make sense that FLASH_UP_EN would be the opposite of CRESET, but there's one problem with it. The I2C-SPI bridge assumes the output pins are used as SS unless configured otherwise, and as such they're all held high by default.

PGOOD is also at 0.0V on boot even it should be high because of that and the PWR GOOD LED will only light up after I switch off FLASH_UP_EN on the I2C-SPI bridge.

gkasprow, 2025-04-25:
It might be the same issue as in Fastino

gkasprow, 2025-04-25:
https://github.com/sinara-hw/Fastino/issues/128

gkasprow, 2025-04-25:
my coleague fixed that by adding one more MOSFET

Spaqin, 2025-04-25:
Yep, seems to be the same problem.

gkasprow, 2025-04-25:
later on I will post fix, ping me in 3 hours :)